ros-industrial/industrial_ci #201: documentation build

asmodehn, 2017-07-31:
Currently industrial_ci is testing build, run_tests, and install target.
It would be useful to include a doc build test, probably following the way the buildfarm does it, or something along the lines of http://wiki.ros.org/rosdoc_lite

ipa-mdl, 2017-07-31:
Do have an example that could be used to test such a feature?

I am not sure if the buildfarm doc scripts could be used like the prerelease script, i.e. to test the merged code.
Running `rosdoc_lite` for each package could be an alternative.

asmodehn, 2017-08-01:
My most documented package is https://github.com/pyros-dev/catkin_pip/tree/industrial_ci. Doc has been generated there previously by the buildfarm : http://docs.ros.org/indigo/api/catkin_pip/html/.

I am currently migrating it to use industrial_ci instead of my custom travis and shell scripts, so we can use that package for testing it.

Using rosdoc_lite should be fine, thats what I usually do locally.

asmodehn, 2017-08-28:
I recently made another package, much simpler, for testing this kind of thing : https://github.com/pyros-dev/ros1_template

artivis, 2017-10-19:
Might be interesting to look at [catkin_tools_document](https://github.com/mikepurvis/catkin_tools_document) a `catkin_tool` plugin for building documentation at workspace-level.

ipa-mdl, 2017-10-23:
Thanks for pointing this out!
I have tried it, but I got stuck with #123 as doxgen writes temporary files to the src folder.
I will give it another try when I have more time.

wxmerkt, 2018-11-22:
We have run into the same issue - beyond doxygen (which writes into the read-only source directory, but can be redirected) we run automatic documentation on a built Python module (bindings from C++), i.e., can only work if that build artefact is available. Is there any example for building documentation or work-around? (I have seen the open PR on making it write-able but it seems to not have been merged/gotten stuck).

agutenkunst, 2021-04-26:
I would like to refresh this issue.

I recently added a rosdoc_lite check to our CI: https://github.com/PilzDE/psen_scan_v2/commit/2d3182212e89f5175323634d97a4ddba65434989

It just validates that no warnings/errors are printed from rosdoc_lite. Unfortunately I had to filter some existing errors that would
need fixes upstream at rosdoc_lite see https://github.com/ros-infrastructure/rosdoc_lite/pull/100

@ipa-mdl would you welcome a check for rosdoc_lite warnings/errors in industrial_ci? 
I think it would fit in with `CATKIN_LINT`, `CLANG_TIDY` and `PYLINT_CHECK`.

ipa-mdl, 2021-04-26:
>  would you welcome a check for rosdoc_lite warnings/errors in industrial_ci?
I think it would fit in with CATKIN_LINT, CLANG_TIDY and PYLINT_CHECK.

Definitely!
I could imagine to have a standalone test for it (similar to https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/tests/black_check.sh).
Something like "tests/doc.sh" (if we want to extend it later) or "tests/rosdoc_lite.sh" ?