ros-industrial/industrial_ci #204: Stalled travis jobs

ipa-mdl, 2017-07-31:
The number of jobs (here and in other repositories) that hit the travis stall detection timeout (default: 10 minutes) increased.

Example:
https://travis-ci.org/ros-industrial/industrial_ci/jobs/259056635 errored, but https://travis-ci.org/ros-industrial/industrial_ci/jobs/258793803 passed (in 3 minutes), both built code for the same commit.

Looks like something got stuck, but I haven't found the problem yet.
SInce our code is unchanged, there might be some change upstream that does work properly with our scripts.

ipa-mdl, 2017-08-01:
Today it went [fine](https://travis-ci.org/ros-industrial/industrial_ci/jobs/259705831)

130s, 2017-12-13:
Can this be closed? Might #205 be taking effect?

130s, 2017-12-13:
Wait, I don't know if it's the same issue or not, but in one of our private repos on gitlab.com CI gets stuck at (seemingly) random place and finish with hitting 1 hour limit. I haven't seen jobs on Gitlab CI halt because of detected inactivity so the observation-wise it sounds pretty similar with what's reported here.

In our case it has always happened during `catkin run_tests` that normally takes 10 to maybe 25 mins (depending on the test case to be used). Some jobs passed but rarely.

ipa-mdl, 2017-12-13:
I would guess that travis fixed something, I know they had some issues as well.
I haven't restarted any job in weeks, so this can be closed.

> Wait, I don't know if it's the same issue or not, but in one of our private repos on gitlab.com CI gets stuck at (seemingly) random place and finish with hitting 1 hour limit.

If you can share any details, please open a new issue for this.

130s, 2017-12-17:
Ok, closing and I'll open a new issue as the issue I'm seeing is on Gitlab CI.

closed 2017-12-17 by 130s