ros-industrial/industrial_ci #206: running tests after install ?

asmodehn, 2017-08-01:
For https://github.com/pyros-dev/catkin_pip I need to run some tests after install (as part of that step, the same way that the build has tests).

This help ensure the install has been done properly (too often people forget to install a script, a launch file, or something else...)

I couldn't find in the documentation how to do it...

ipa-mdl, 2017-08-01:
You could put your additional steps in a script and call it with `AFTER_SCRIPT` (run after all other tests).
If the tests should be run from a catkin package, there is the `USE_MOCKUP` feature. It links in an additional tree of packages that are ignored by default ([example](https://github.com/ros-infrastructure/rosdep)). If the mockup packages depend on your target packages, they should be built/tested of the target package was installed (to be tested).

asmodehn, 2017-08-02:
The documentation says : 

> AFTER_SCRIPT: (default: not set): Used to specify shell commands that run after all source tests. 

But I don't want to run it after source tests, I want to run it after install, after having done the `source install/setup.bash`. This is useful for when your package embed tests to be run on the user environment... my usual travis hceck shell script just looks like : 
```bash
mkdir -p build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=./install
if [ "$ROS_FLOW" == "devel" ]; then
    make -j1
    source devel/setup.bash
    rospack profile
    make -j1 tests
    make -j1 run_tests
    catkin_test_results .
elif [ "$ROS_FLOW" == "install" ]; then
    make -j1 install
    source install/setup.bash
    rospack profile
    # some test packages have embedded tests to validate installed package structure
    python -m nose mypkgtests
fi
```

The `USE_MOCKUP` also doesn't seem to be a good fit, since I don't need any extra package for this, I just want to run code that is already in the installed package...

asmodehn, 2017-08-02:
I think catkin_pip is probably too complex to check this, other differences from a normal ROS package are getting mixed up...

Here is a simple ROS package that has the same structure : https://github.com/pyros-dev/pyros-utils, I would like to implement that feature there first. I started a PR for it : https://github.com/pyros-dev/pyros-utils/pull/7/files

ipa-mdl, 2017-08-02:
> But I don't want to run it after source tests, I want to run it after install, after having done the source install/setup.bash.

"source test" means after the regular ` industiral_ci` run was completed.

`USE_MOCKUP` was implemented to provide packages for our builds to fail for a specific reason that needs to get dteected.

If you want to run nosetests then `"AFTER_SCRIPT=python -m nose mypkgtests"` might just work for you.

asmodehn, 2017-08-05:
Turns out I need to source the install space myself doing : 
`AFTER_SCRIPT="cd ~/catkin_ws && source install/setup.bash && python -m nose pyros_utils"`
You can see the result (package building finally) there : https://travis-ci.org/pyros-dev/pyros-utils/builds/261293568
I found out `sources_test.sh` was using this `~/catkin_ws`, so I decided to use the same.

This took me way much more time than I would have thought, which is why I think some kind of simplification is definitely needed. See : https://github.com/pyros-dev/pyros-utils/pull/7
Honestly, currently copy-pasting my 2 small files, and customising them is still faster than digging in industrial_ci scripts to find how things work and what I need to set...

We can use this package for testing any kind of "install space tests" feature you d like, it is simple enough. Just let me know when I can help with it.

ipa-mdl, 2017-08-05:
Simpler version: `AFTER_SCRIPT="$CATKIN_WORKSPACE/install/env.sh python -m nose pyros_utils"`.

asmodehn, 2017-08-11:
Doesn't seem to work : https://travis-ci.org/pyros-dev/pyros-utils/jobs/262968598
Apparently $CATKIN_WORKSPACE is not known when running the after_script...

ipa-mdl, 2017-08-11:
> Doesn't seem to work

Please try single quotes instead.

> Apparently $CATKIN_WORKSPACE is not known when running the after_script...

It is known, but the string gets substituted before running 'industrial_ci`.

asmodehn, 2017-08-11:
Ah yeah I see https://github.com/travis-ci/travis-ci/issues/2477
OK seems fine now : https://travis-ci.org/pyros-dev/pyros-utils/builds/263399141

ipa-mdl, 2017-10-28:
This seems to be resolved.

130s, 2017-11-04:
Reopening as a reminder to add more document for `USE_MOCKUP` (which just saved me now).

ipa-mdl, 2019-05-15:
> Reopening as a reminder to add more document for USE_MOCKUP (which just saved me now).

`USE_MOCKUP` gets deprecated/removed in #361

closed 2019-05-15 by ipa-mdl