ros-industrial/industrial_ci #207: running tests from custom command ?

asmodehn, 2017-08-01:
This is because in https://github.com/pyros-dev/catkin_pip I have : 
- a `CMakeLists.txt` to build the package (on the rosbuildfarm, cleanly)
- a `CMakeLists.txt` torun all sorts of tests (requiring dependencies not on teh buildarm) that I want to run on travis only, but not on the buildfarm, so they have to be separate.

How can I run a custom command for testing my build ?

ipa-mdl, 2017-08-01:
https://github.com/ros-industrial/industrial_ci/issues/206#issuecomment-319283643 applies here as well.

asmodehn, 2017-08-02:
The `USE_MOCKUP` is probably a good fit here...

I implemented a similar feature with cmake and a symlink, so that travis runs a CMakeLists with a bunch of packages for testing, while the buildfarm will run another CMakeLists.

It looks like in industrial_ci, it is done by manually running CMakelists of packages that are usually ignored... I ll work on restructuring catkin_pip package in a similar way, I ll report back here when I m done.

asmodehn, 2017-08-02:
After some more investigation, In that package I also want to run tests between packages (check if their pythonpath is set properly, etc.). You can check that here : https://github.com/pyros-dev/catkin_pip/blob/devel/test/CMakeLists.txt#L82

I m thinking that might be too much flexibility to ask from industrial_ci, and I should go back to doing it directly with CMake where I can customize everything... It might be better for a package involved in the build system itself like catkin_pip...

ipa-mdl, 2017-08-02:
As long as you can put in in a `CMakeLists.txt` it should get tested by `run_tests`.
For more complex tests you have to write an `AFTER_SCRIPT`.
I plan to implement custom tests scripts later.

ipa-mdl, 2019-05-15:
> I plan to implement custom tests scripts later.

New hook system gets introduced in #361

ipa-mdl, 2021-09-16:
> I plan to implement custom tests scripts later.

External tests can now be provided from custom files: `TEST=my_pkg/my_test.sh`

closed 2021-09-16 by ipa-mdl