ros-industrial/industrial_ci #208: Access output log on travis ?

asmodehn, 2017-08-02:
When a build fail on travis I get something like : 
```
----------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
Starting  >>> catkin_tools_prebuild               
Finished  <<< catkin_tools_prebuild                [ 1.2 seconds ]
Starting  >>> ros1_pip_template                   
_______________________________________________________________________________
Errors     << ros1_pip_template:make /root/catkin_ws/logs/ros1_pip_template/build.make.000.log
Command "python setup.py egg_info" failed with error code 1 in /root/catkin_ws/src/ros1-pip-template/
[...]
make[2]: *** [/root/catkin_ws/devel/.private/ros1_pip_template/lib/python2.7/site-packages/ros1_pip_template.egg-link] Error 1
make[1]: *** [CMakeFiles/ros1_pip_template.dir/all] Error 2
make: *** [all] Error 2
```

So all the information about that build is in the build.make.000.log

- is there a way to access this file on travis that I am not aware of ?
- why not output everything directly on the same travis log output ?

Without this, actually figuring out what the error is, requires me to run the same locally and hope it break in the same way. Maybe I m missing something ?

ipa-mdl, 2017-08-02:
> is there a way to access this file on travis that I am not aware of ?

You could mount the log folder and output it, but I would not recommend this.

> why not output everything directly on the same travis log output ?

There is a limit of 10.000 lines that can be displayed by travis, and there is an maximum limit logs in travis.
You could try so set `CATKIN_VERBOSE=true`.

Perhaps we can add a feature to dump the logs is catkin fails..

asmodehn, 2017-08-03:
`VERBOSE_OUTPUT=true` did the trick indeed. log is displayed as I am used to with catkin_make.
Thanks !

closed 2017-08-03 by asmodehn