ros-industrial/industrial_ci #212: ROS distro '' not supported with DOCKER_IMAGE

asmodehn, 2017-08-04:
Setting up to use the DOCKER_IMAGE, I get : 

```
**Setting environment variables from .travis.yml**
$ export VERBOSE_OUTPUT=true
$ export DOCKER_IMAGE=indigo-trusty-shadow-fixed
$ bash -c 'echo $BASH_VERSION'
4.3.11(1)-release

$ git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
Cloning into '.ci_config'...
remote: Counting objects: 1479, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1479 (delta 0), reused 2 (delta 0), pack-reused 1475
Receiving objects: 100% (1479/1479), 431.98 KiB | 0 bytes/s, done.
Resolving deltas: 100% (934/934), done.

$ .ci_config/travis.sh
ROS distro '' is not supported
```

But ROS distro should not be needed since I specified DOCKER_IMAGE.

asmodehn, 2017-08-04:
Ref : https://travis-ci.org/asmodehn/pyros-test/jobs/260854793

ipa-mdl, 2017-08-04:
> But ROS distro should not be needed since I specified DOCKER_IMAGE.

This does not hold for all images, e.g. the `ros:*` images don't set this.
We could try to set them in our images.

130s, 2018-12-20:
Addressed by https://github.com/ros-industrial/industrial_ci/pull/280

closed 2018-12-20 by 130s