ros-industrial/industrial_ci #213: gitlab-ci job failed

vincentrou, 2017-09-07:
I try to use insdustrial_ci on gitlab instead of my homemade scripts.
But the job failed. It tries to execute a file named /root/ici/ci_main.sh but it does not exist.
This path comes from here https://github.com/ros-industrial/industrial_ci/blob/6dfcee82109157888fb93a6619be4022a73162ed/industrial_ci/src/docker.sh#L40 but I don't know what is this folder named "/root/ici".
Here is the log.

```
Running with gitlab-ci-multi-runner 9.5.0 (413da38)
  on CFP0196 (01fff31c)
Using Docker executor with image docker:git ...
Starting service docker:dind ...
Pulling docker image docker:dind ...
Using docker image docker:dind ID=sha256:7731e67e3def0af852ea8ab3788063fc640ce359d399b86fe5a009e6d814f625 for docker service...
Waiting for services to be up and running...
Using docker image sha256:0b3f8c0899237f2dd669ed3e3ef941c90fbd30fb5129653121890336bd72f650 for predefined container...
Pulling docker image docker:git ...
Using docker image docker:git ID=sha256:9fbf23232776ff80a3e098c2f54794e0b581e9dccf81a1bbc0349ab5baf0d220 for build container...
Running on runner-01fff31c-project-318-concurrent-0 via CFP0196...
Fetching changes...
Removing .ci_config/
HEAD is now at dc8403f Add gitlab ci from ros industrial
Checking out dc8403f9 as master...
Skipping Git submodules setup
$ apk add --update bash coreutils
fetch http://dl-cdn.alpinelinux.org/alpine/v3.6/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.6/community/x86_64/APKINDEX.tar.gz
(1/8) Installing ncurses-terminfo-base (6.0-r8)
(2/8) Installing ncurses-terminfo (6.0-r8)
(3/8) Installing ncurses-libs (6.0-r8)
(4/8) Installing readline (6.3.008-r5)
(5/8) Installing bash (4.3.48-r1)
Executing bash-4.3.48-r1.post-install
(6/8) Installing libattr (2.4.47-r6)
(7/8) Installing libacl (2.2.52-r3)
(8/8) Installing coreutils (8.27-r0)
Executing busybox-1.26.2-r5.trigger
OK: 41 MiB in 27 packages
$ git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
Cloning into '.ci_config'...
$ .ci_config/gitlab.sh
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Building image 'industrial-ci/kinetic/ubuntu:xenial':
FROM ubuntu:xenial

RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections

RUN apt-get update -qq     && apt-get -qq install --no-install-recommends -y apt-utils gnupg wget ca-certificates sudo lsb-release

RUN echo "deb http://packages.ros.org/ros-shadow-fixed/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list
RUN apt-key adv --keyserver "hkp://ha.pool.sks-keyservers.net" --recv-key "0xB01FA116"     || { wget "https://raw.githubusercontent.com/ros/rosdistro/master/ros.key" -O - | sudo apt-key add -; }

RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-kinetic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function prepare_docker_image returned with code '0' after 0 min 0 sec 
/bin/bash: /root/ici/ci_main.sh: No such file or directory
ERROR: Job failed: exit code 127
```

ipa-mdl, 2017-09-07:
> /bin/bash: /root/ici/ci_main.sh: No such file or directory

This happens if something is wrong with the docker mounts.
Gitlab runners do not expose all directories, I have learned that the hard way :-/

Which runner do you use?
What's your CI config, runner config?

The [example config](https://github.com/ros-industrial/industrial_ci/blob/master/.gitlab-ci.yml) works for the public shared runners ([example pipeline](https://gitlab.com/ipa-mdl/ci-example/pipelines/11550890)).

vincentrou, 2017-09-08:
Yes it should be something like that. I don't know the config.toml file for the public runners of gitlab.com...
I am not sure this is updated : https://about.gitlab.com/2016/04/05/shared-runners/
I have installed the gitlab-ci-multi-runner package on my ubuntu desktop.
here is my config.toml file :
```
concurrent = 1
check_interval = 0

[[runners]]
  name = "CFP0196"
  url = "https://gitlab.xxxxxx.fr/ci"
  token = "TOKEN"
  executor = "docker"
  [runners.docker]
    tls_verify = false
    image = "gcc"
    privileged = true
    disable_cache = false
    volumes = ["/var/run/docker.sock:/var/run/docker.sock", "/cache"]
  [runners.cache]
```
my .gitlab-ci.yml file :
```
image: docker:git
variables:
  DOCKER_DRIVER: "overlay" # recommended for docker-in-docker
services:
  - docker:dind # enable docker-in-docker
before_script:
  - apk add --update bash coreutils # install travis depedencies
  - git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
kinetic:
  script: 
    - .ci_config/gitlab.sh
  variables:
    ROS_DISTRO: "kinetic"
```

ipa-mdl, 2017-09-12:
You runner shares the docker socket, so you don't have to use DinD service.
Please try to comment this out.
I am not sure about the `DOCKER_DRIVER` setting, depending on your set-up this might not work.

miguelprada, 2017-09-13:
> You runner shares the docker socket, so you don't have to use DinD service.

I believe you do need DinD for the shared volumes. It's the runner who downloads both the project source and the industrial_ci code, but if you use the shared docker socket then docker will try to mount the volumes from the host, where neither the project source nor the industrial_ci sripts are present.

ipa-mdl, 2017-09-13:
>  but if you use the shared docker socket then docker will try to mount the volumes from the host, where neither the project source nor the industrial_ci sripts are present.

This makes sense.
I will try to replicate it with my gitlab instance.

vincentrou, 2017-09-14:
I have the same problem when I remove those lines in the .gitlab-ci.yml :
```
variables:
  DOCKER_DRIVER: "overlay" # recommended for docker-in-docker
services:
  - docker:dind # enable docker-in-docker
```

ipa-mdl, 2017-09-14:
It works with my local instance, except for the docker-in-docker-in-docker pre-release test..

My config is almost the same, but I have set `image = "docker:latest"`.
However I guess that this will be overwritten anyway (.gitlab-ci.yml).

```
Running with gitlab-ci-multi-runner 9.3.0 (3df822b)
  on ***
Using Docker executor with image docker:git ...
Starting service docker:dind ...
Pulling docker image docker:dind ...
Using docker image docker:dind ID=sha256:1ab52967778e46d72867aa7cc962fa5aff6fc2fa56cf0564d49e392531853f7a for docker service...
Waiting for services to be up and running...
Using docker image sha256:102b41d50bece7595ea38db5d7a3e17936c98a8d210893b1b5d89a3102b8ae31 for predefined container...
Pulling docker image docker:git ...
Using docker image docker:git ID=sha256:495a38c313233cf316b6ec3cc6e3eb00e0a329a30cbf7354d396b388f2086cdc for build container...
```

Apart from the older runner, my `docker:dind` and `docker:git` are outdated as well.
I will update them until they fail ;)

ipa-mdl, 2017-09-14:
@vincentrou: Actually, my images are up-to-date.

ipa-mdl, 2017-09-14:
With #214 now all example jobs run on my updated runner.

```
Running with gitlab-ci-multi-runner 9.5.0 (413da38)
  on ***
Using Docker executor with image docker:git ...
Starting service docker:dind ...
Pulling docker image docker:dind ...
Using docker image docker:dind ID=sha256:1ab52967778e46d72867aa7cc962fa5aff6fc2fa56cf0564d49e392531853f7a for docker service...
Waiting for services to be up and running...
Using docker image sha256:b8f9ec77791fb499cb6961f23a26f913bda4c775322943f9ce6ecabb0b003d49 for predefined container...
Pulling docker image docker:git ...
Using docker image docker:git ID=sha256:495a38c313233cf316b6ec3cc6e3eb00e0a329a30cbf7354d396b388f2086cdc for build container...
```
```
# docker --version                                                                                                                                                                                                     
Docker version 17.05.0-ce, build 89658be
```

ipa-mdl, 2017-10-28:
@vincentrou: do you still have this problem?

vincentrou, 2017-10-30:
I still have problems. The problem is when I use specific runners.
Because I try to reproduce it on gitlab.com.
When using shared runners it works : https://gitlab.com/vincentrou/industrial_ci/pipelines/13352664
But when I use specific runners (my computer) it fails : https://gitlab.com/vincentrou/industrial_ci/pipelines/13361938

Maybe it comes from from my /etc/gitlab-runner/config.toml

```
concurrent = 1
check_interval = 0

[[runners]]
  name = "CFP0196"
  url = "https://gitlab.com"
  token = "TOKEN"
  executor = "docker"
  [runners.docker]
    tls_verify = false
    image = "docker:latest"
    privileged = true
    disable_cache = false
    volumes = ["/var/run/docker.sock:/var/run/docker.sock", "/cache"]
    shm_size = 0
  [runners.cache]

```

ipa-mdl, 2017-10-30:
Thanks for the update!

AFAIR I have the same config, but I will double check when I am in the office again.
I cannot remember if I have removed  `DOCKER_DRIVER: "overlay"` in my local test. 

I would suggest to inspect your local docker containers to get a hint on what's going on.

vincentrou, 2017-11-03:
Tanhks @ipa-mdl , I made it work by removing the `DOCKER_DRIVER: "overlay"` line.
https://gitlab.com/vincentrou/industrial_ci/commit/e4cba05650b07563d0fb4ff988d2d3317d0f2269

We should remove this line from the default [example file](https://github.com/ros-industrial/industrial_ci/blob/master/.gitlab-ci.yml) since it does not work with specific runner right ?

ipa-mdl, 2017-11-03:
> We should remove this line from the default example file since it does not work with specific runner right ?

I will run a test with the shared runners first.
The least we can do is to point out that this setting does not work for most (?) runners.

vincentrou, 2017-11-03:
I have rerun with the shared runners and it also pass
https://gitlab.com/vincentrou/industrial_ci/pipelines/13539028

ipa-mdl, 2017-11-04:
I have run some futher tests: [overlay](https://gitlab.com/ipa-mdl/industrial_ci/-/jobs/39001483), [overlay2](https://gitlab.com/ipa-mdl/industrial_ci/-/jobs/39001551), [no overlay](https://gitlab.com/ipa-mdl/industrial_ci/-/jobs/39001569).
So at least for the shared runners this does not make a noticeable difference.

I will update the example config and add a link to https://docs.gitlab.com/ce/ci/docker/using_docker_build.html#using-the-overlayfs-driver

ipa-mdl, 2017-11-04:
done in #226

PR #226 (2017-11-11): remove overlay driver setting from gitlab example
It might break local builds if the overlay driver is not enabled in the kernel.
Instead a link to the Gitlab documentation was added.

closes #213
closed 2017-11-11 by 130s