ros-industrial/industrial_ci #216: [gitlab] Error "/usr/bin/catkin: No such file or directory"

130s, 2017-09-21:
In our private repo I started seeing this error after I switched to using `ros:indigo-ros-core` image.
```
:
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
Add distro "lunar"
updated cache in /root/.ros/rosdep/sources.cache
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function setup_rosdep returned with code '0' after 0 min 17 sec 
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Writing /root/catkin_ws/src/.rosinstall

update complete.
Obtain deb binary for upstream packages.
/root/ici/tests/source_tests.sh: line 44: /usr/bin/catkin: No such file or directory
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function setup_rosws returned with code '1' after 0 min 0 sec 
ERROR: Job failed: exit code 1
```

The entire `.gitlab.yaml`:
```
# the following block is needed for the shared Docker-based runner
image: docker:git
variables:
  DOCKER_DRIVER: "overlay" # recommended for docker-in-docker
services:
  - docker:dind # enable docker-in-docker
before_script:
  - apk add --update bash coreutils  # install industrial_ci depedencies
  - git clone https://github.com/ros-industrial/industrial_ci .ci_config
indigo:
  script:
    - .ci_config/gitlab.sh
  variables:
    ROS_DISTRO: "indigo"
    DOCKER_IMAGE: "ros:indigo-ros-core"
```

I think when `DOCKER_IMAGE` is set, [simply the image is pulled](https://github.com/ros-industrial/industrial_ci/blob/6dfcee82109157888fb93a6619be4022a73162ed/industrial_ci/src/docker.sh#L150) so whether `python-catkin-tools` is available depends on the image?

ipa-mdl, 2017-09-21:
> I think when DOCKER_IMAGE is set, simply the image is pulled so whether python-catkin-tools is available depends on the image?

Exactly!
The `DOCKER_IMAGE` is meant to already include everthing we need.
You can [make it work](https://github.com/ros-industrial/industrial_ci/blob/master/.travis.yml#L28) with the official ROS images as well.

As an alternative you could use `rosindustrial/ci:indigo-trusty`.

130s, 2017-09-22:
Thanks @ipa-mdl 
I added in https://github.com/ros-industrial/industrial_ci/pull/215 some improvement regarding the situation like this.

closed 2017-09-22 by 130s