ros-industrial/industrial_ci #217: [docker-in-docker] Location of script files for {BEFORE, AFTER}_SCRIPT

130s, 2017-09-22:
I think the [usage of `{BEFORE, AFTER}_SCRIPT`}(https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#run-pre-post-process-custom-commands) was written before we introduced Docker-in-Docker and thus outdated. Now I get `bash: %SCRIPTNAME%: command not found` error (e.g. [in this line](https://travis-ci.org/ros-industrial/industrial_calibration/jobs/278522903#L597) for [this PR](https://github.com/ros-industrial/industrial_calibration/pull/112), particularly [this commit](https://github.com/ros-industrial/industrial_calibration/pull/112/commits/536014b84bdcc078d009eaa2982e2ca624533c94), unless I'm doing something wrong there). 

What the right path of the scripts to pass to these variables?

ipa-mdl, 2017-09-22:
Please try `./.travis_ceres.sh`, otherwise it get looked up in `PATH`.

130s, 2017-09-23:
Ah...that was my carelessness. Doc specifies `./%SCRIPT%` path. Thanks.

closed 2017-09-23 by 130s