ros-industrial/industrial_ci #218: how to fix dependency issues

mauriceAlthoff, 2017-09-29:
Hey I don't know if this is the right place to ask questions about using industrial_ci bud here is my problem:

I try to get industrial-ci to run on travis-ci and get stuck with dependency issues. On my local machine I can just install the missing packages bud the travis script seams not be able to handle that.
Here is the link to the Build Log:
[error msg](https://travis-ci.org/mauriceAlthoff/modelcars/builds/281210070)

can someone please help me?

Greetings maurice

gavanderhoorn, 2017-09-29:
I think the problems is actually that `opencv2` is not an actual `rosdep` key:

```
RuntimeError: Could not resolve the rosdep key 'opencv2'
```

The only `rosdep` key for opencv is `libopencv-dev` afaict.

https://github.com/ros/rosdistro/blob/5ea2c2885aa3690846c4876dd784203b5ec06765/rosdep/base.yaml#L2170-L2182

ipa-mdl, 2017-09-29:
@gavanderhoorn is right, you have to find the proper keys for your depedencies.

> On my local machine I can just install the missing packages 

Are you installing it with `rosdep` or `apt-get`?
You can test it locally with `rosdep install -i --from-paths PATH_TO_YOUR_REPO`

mauriceAlthoff, 2017-09-29:
thanks, I could fix the errors

closed 2017-09-29 by mauriceAlthoff