ros-industrial/industrial_ci #223: How to access to private repo via ssh is unclear

130s, 2017-10-29:
Assigning myself. Likely this will end up adding documentation at most.

I'm still struggling after getting some advices in https://github.com/ros-industrial/industrial_ci/pull/220 from @ipa-mdl to `git clone` from a repo to a private repo on gitlab.

Using sample projects, From [sample_pub_repo](https://gitlab.com/130s/sample_pub_repo/blob/master/.gitlab-ci.yml) I'm trying to `git clone` [sample_private_repo](https://gitlab.com/130s/sample_private_repo) where a pub key is deployed. 

https://gitlab.com/130s/sample_pub_repo/-/jobs/38093210
```
Cloning into 'sample_private_repo'...
Warning: Permanently added 'gitlab.com,52.167.219.168' (ECDSA) to the list of known hosts.

Permission denied (publickey).

fatal: Could not read from remote repository.
```

Using a temporary branch of industrial_ci to print debug messages in Docker, `~/.ssh/known_hosts` being empty seems critical to me.
```
DEBUG1 /tmp/ssh-6jyYxQbE2C9G/agent.27
DEBUG2 .ssh copied
total 32
drwxr-xr-x 2 root root 4096 Oct 29 01:59 .
drwx------ 1 root root 4096 Oct 29 01:59 ..
-rw-r--r-- 1 root root   35 Oct 29 01:59 config
-rw-r--r-- 1 root root    1 Oct 29 01:59 known_hosts

```

miguelprada, 2017-12-04:
I got it working in a less-than-stellar fashion. In the before script of my `.gitlab-ci.ym` I usually add:

```
before_script:
  ...
  - mkdir -p ~/.ssh
  - echo "$SSH_KEY_SECRET" > ~/.ssh/id_rsa
  - chmod og-r ~/.ssh/id_rsa
  - echo -e "Host my.gitlab.host.com\n\tStrictHostKeyChecking no\n\n" > ~/.ssh/config
```

This is being run by a Docker executor based on `docker:git` and using `docker:dind` as a service.

I do feel that there must be a better way, though.

ipa-mdl, 2017-12-04:
Has anyone of you actually tried [this](https://github.com/ros-industrial/industrial_ci/pull/220#issuecomment-340076453)?

It has two advantanges:
1. The private key will not get revealed
2.  Known host keys have to be provided, so no need to facilitate man-in-the-middle attacks

130s, 2017-12-28:
I think this is addressed by https://github.com/ros-industrial/industrial_ci/pull/243

closed 2017-12-28 by 130s