ros-industrial/industrial_ci #224: CRON job failing

130s, 2017-10-29:
The last 2 CRON jobs seem to have been failed by the same job: [ROS_DISTRO=kinetic _EXTERNAL_REPO='ipa320/cob_calibration_data#c0922cc6f87f31641add6ba4e0fb88c93f9a7c6c' ROS_REPO=ros UPSTREAM_WORKSPACE=file AFTER_SCRIPT='./.travis.xacro_test.sh'](https://github.com/ros-industrial/industrial_ci/blob/33be4ab4e1bc6bc9ca869b2bff11b3447b2ffe1b/.travis.yml#L64)

Output on Travis CI. https://travis-ci.org/ros-industrial/industrial_ci/jobs/294409361#L686
```
Starting  >>> cob_calibration_data                
_______________________________________________________________________________
Errors     << cob_calibration_data:install /root/catkin_ws/logs/cob_calibration_data/build.install.000.log
CMake Error at cmake_install.cmake:169 (file):
  file INSTALL cannot find
  "/root/catkin_ws/src/cob_calibration_data/cob4-16".
make: *** [install] Error 1
```
I skimmed through the upstream but couldn't even find the targeted commit. @ipa-mdl Do you have any idea since you added this https://github.com/ros-industrial/industrial_ci/commit/0d6c5b5b1dfea306b257f3a614ac803b2f4b6b84#diff-354f30a63fb0907d4ad57269548329e3?

ipa-fxm, 2017-10-29:
sorry, that was me...
we added a new robot `cob4-16` in ipa320/cob_supported_robots#12, but ipa320/cob_calibration_data#143 has not been merged yet
unfortunately, I can't merge without one of ipa320's owners approval

ipa-mdl, 2017-10-29:
> sorry, that was me...

Not entirely ;)

This package depends on https://github.com/ipa320/cob_supported_robots and its list of robots.
We should switch back to using `indigo_dev` and wait for the problem to settle.

PR #229 (2017-11-04): fix external test
fixes #224
closed 2017-11-04 by 130s