ros-industrial/industrial_ci #232: Running CI on bare rosinstall repo

miguelprada, 2017-11-16:
I'm recently using a workflow which consists in having repos with only a rosinstall file which I then use to initialize and manage my workspace. Now I'm experimenting a bit with adding industrial_ci config to those repos as well (I know this might seem odd but bear with me, please).

I'm stuck in the catkin_run_tests step. Here, the tests are run on the packages listed in `PKGS_DOWNSTREAM`, which gets filled with `TARGET_PKGS`, which in turn gets filled with the contents in `TARGET_REPO_PATH`. My problem here is that there's no package at all there, only a rosinstall file.

At first I thought I could dynamically fill either `PKGS_DOWNSTREAM` or `TARGET_PKGS` in the before_script step, but since this is run in a sub-shell I cannot modify the parent's shell environment from there.

Then I tried some crazier ideas of assigning some substitution command to either of those variables and hoping them to be evaluated at the right time, but I failed miserably.

Has anyone attempted anything similar or has anyone any idea of how I could achieve this? Almost anything is better than hardcoding `PKGS_DOWNSTREAM` with all the package names in my CI config file...

ipa-mdl, 2017-11-16:
For the sake of speeding up the tests, we don't run the tests for the upstream packages..
I would even consider it a bug that the upstream packages get built ;)

The `--no-deps` flag is breaking the test build.
We might consider removing it, because it prevents the dependencies of the downstream packages from being built if `BUILD_PKGS_WHITELIST` is set.

miguelprada, 2017-11-20:
> For the sake of speeding up the tests, we don't run the tests for the upstream packages..
I would even consider it a bug that the upstream packages get built ;)

Makes perfect sense.

> The --no-deps flag is breaking the test build.

Indeed, I just tested without that flag and it works perfectly.

> We might consider removing it, because it prevents the dependencies of the downstream packages from being built if BUILD_PKGS_WHITELIST is set.

Would you consider a PR adding an additional `TEST_UPSTREAM` environment variable which removes this flag when set to `true`?

ipa-mdl, 2017-11-20:
> Would you consider a PR adding an additional TEST_UPSTREAM environment variable which removes this flag when set to true?

Only as a last escape.
For now I would opt for removing the flag completely.

miguelprada, 2017-12-04:
> For now I would opt for removing the flag completely.

But in that case all test for upstream packages will get built and run, right? You sure you want to push this change to all packages relying on `industrial_ci`?

I have fixed my issue by setting `NOT_TEST_BUILD` in my job configurations, which is good enough for my current scenario. I can either close this issue or open a PR removing the `--no-deps` flag altogether. Which one would you prefer?

ipa-mdl, 2017-12-04:
If you set `NOT_TEST_BUILD` the test won't be run(?).
I prefer removing `--no-deps`, as it might be error-prone for certain downstream setups.

miguelprada, 2017-12-05:
> good enough for my current scenario.

I am relying on several internal packages which have not set up CI yet, let alone tests. This is my way of at least making sure that they all build together in a clean environment.

ipa-mdl, 2017-12-13:
@miguelprada: Does it work for you now? Can we close this?

miguelprada, 2017-12-14:
Works like a charm. Thanks @ipa-mdl!

closed 2017-12-14 by miguelprada