ros-industrial/industrial_ci #235: Add support for ARM-based containers and Debian

ahoarau, 2017-11-24:
Now that Docker supports ARM : 
https://discourse.ros.org/t/announcing-ros-docker-images-for-arm-and-debian/2467

That'd be a very useful feature to have in i-ci !

ipa-mdl, 2017-11-24:
If  I am not mistaken, it is already supported, but it takes additional configuration options.
I'll try it and report back to you :)

ipa-mdl, 2017-11-24:
It works somehow (but needs #236):
```yml
env:
  matrix:
    - ROS_DISTRO=lunar DOCKER_IMAGE='arm32v7/ros:lunar'  ADDITIONAL_DEBS="build-essential python-catkin-tools python-pip" DOCKER_RUN_OPTS="-v /usr/bin/qemu-arm-static:/usr/bin/qemu-arm-static"
script:
  - sudo apt-get install qemu-user-static
  - industrial_ci/_wrap_test.sh # this script is only for internal use
```
The tricky part is to inject qemu into the image.
I will try to add support for our default images.

ahoarau, 2017-11-24:
excellent ! That'd be great if we could have an option like `ARCH=arm64v8` to setup all of these

ipa-mdl, 2017-11-25:
> That'd be great if we could have an option like ARCH=arm64v8 to setup all of these

#237 implements automatic QEMU injection.
It is not as simple as `ARCH=arm64v8`, but much more flexible and better to maintain.
In order to cross-build, you should configure an ARM-based image and enable QEMU-injection.
However, cross-building on Travis is really slow (factor 10!), so it would be better to prepare a dedicated ARM-based image (including QEMU) for speeding up the CI tests.

130s, 2017-12-13:
Is this ready to be closed, or do we still need #237 etc.?

ipa-mdl, 2017-12-13:
I would keep it open until #237 is closed.

closed 2017-12-31 by 130s

lsolanka-flya, 2019-10-23:
It looks like the support for this does not work on Bitbucket due to not being able to mount into the docker container from `/usr/bin/qemu-${ARCH}-static`.

Do you have any suggestions or working solution for BitBucket?

ipa-mdl, 2019-10-23:
> Do you have any suggestions or working solution for BitBucket?

DId you install `qemu-user-static` on the host?

lsolanka-flya, 2019-10-24:
> > Do you have any suggestions or working solution for BitBucket?
> 
> DId you install `qemu-user-static` on the host?

Yes, indeed I did. This is the pipeline file that I used:
```
image: atlassian/default-image:2

pipelines:
  default:
    - parallel:     
      - step:
          name: Test build amd64
          services:
            - docker
          script:
            - apt-get update && apt-get install -y bash coreutils tar
            - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
            - .industrial_ci/bitbucket.sh ROS_DISTRO=kinetic DEBUG_BASH=true DOCKER_IMAGE=ros:kinetic-perception
      - step:
          name: Test build aarch64
          services:
            - docker
          script:
            - apt-get update && apt-get install -y bash coreutils tar qemu-user-static
            - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
            - .industrial_ci/bitbucket.sh ROS_DISTRO=kinetic DEBUG_BASH=true DOCKER_IMAGE=arm64v8/ros:kinetic-perception INJECT_QEMU=aarch64
```
The log file in the pipeline shows 2 issues:
* when installing qemu-user-static `binfmt-support` and `qemu-user-static` print a few warnings:
```
Setting up binfmt-support (2.1.6-1) ...
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of start.
Setting up qemu-user-static (1:2.5+dfsg-5ubuntu10.41) ...
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
update-binfmts: warning: Couldn't load the binfmt_misc module.
Processing triggers for libc-bin (2.23-0ubuntu11) ...
Processing triggers for systemd (229-4ubuntu21.21) ...
```

* Docker commands that mount the qemu static binary show the following error:
```
++ docker create --env-file /opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src/docker.env -v /usr/bin/qemu-aarch64-static:/usr/bin/qemu-aarch64-static:ro -e TARGET_REPO_PATH=<redacted> -v /opt/atlassian/pipelines/agent/build/:<redacted>:ro -v /opt/atlassian/pipelines/agent/build/.industrial_ci/industrial_ci/src/:/root/ici:ro -t --entrypoint '' -w <redacted> arm64v8/ros:kinetic-perception /bin/bash /root/ici/ci_main.sh
Error response from daemon: authorization denied by plugin pipelines: -v only supports $BITBUCKET_CLONE_DIR and its subdirectories
```

This seems to be related to the security restriction on bitbucket pipelines that doesn't allow to mount anything outside `$BITBUCKET_CLONE_DIR` to the guest container: https://confluence.atlassian.com/bitbucket/run-docker-commands-in-bitbucket-pipelines-879254331.html

So I tried the following step, to mount qemu-aarch64-static directly from `$BITBUCKET_CLONE_DIR`:
```
      - step:
          name: Test build aarch64
          services:
            - docker
          script:
            - apt-get update && apt-get install -y bash coreutils tar qemu-user-static
            - apt-get download qemu-user-static && dpkg -x qemu-user-static*.deb $BITBUCKET_CLONE_DIR/qemu/ && ls $BITBUCKET_CLONE_DIR/qemu/usr/bin/qemu-aarch64-static
            - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
            - .industrial_ci/bitbucket.sh ROS_DISTRO=kinetic DEBUG_BASH=true DOCKER_IMAGE=arm64v8/ros:kinetic-perception DOCKER_RUN_OPTS="-v $BITBUCKET_CLONE_DIR/qemu/usr/bin/qemu-aarch64-static:/usr/bin/qemu-aarch64-static:ro"
```
But when running the docker container I get the following error:
```
docker run --rm -v /opt/atlassian/pipelines/agent/build/qemu/usr/bin/qemu-aarch64-static:/usr/bin/qemu-aarch64-static:ro --entrypoint '' arm64v8/ros:kinetic-perception id -u
standard_init_linux.go:207: exec user process caused "exec format error"
```

It looks like it is not enough to just mount the qemu static binary but `binfmt_misc` must be registered in the kernel to make the container run.

Hope this helps to get an idea of where the problem is.

ipa-mdl, 2019-10-24:
I see. The same problem should occur with Gitlab :-/

> It looks like it is not enough to just mount the qemu static binary but binfmt_misc must be registered in the kernel to make the container run.

Yes, and it the binary must be in the same place on the host and in the image.
Starting with Ubuntu Cosmic or Debian Buster (on the host), this should not be needed anymore.
(https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=868030)

Please try
```
DOCKER_BASE_IMAGE=arm64v8/ros:kinetic-perception INJECT_QEMU=aarch64
```
This is not a proper fix, but a proof-of-concept.
It will inject qemu into the image instead of bind-mounting it.

lsolanka-flya, 2019-10-24:
This does not seem to make a difference, the following step still can't run in the pipeline:
```
image: lsolankaflya/atlassian-default-image-qemu:19.04

pipelines:
  default:
    - parallel:     
      - step:
          name: Test build aarch64
          services:
            - docker
          script:
            - apt-get update && apt-get install -y bash coreutils tar qemu-user-static git
            - git clone --quiet --depth 1 https://github.com/ros-industrial/industrial_ci .industrial_ci -b master
            - .industrial_ci/bitbucket.sh ROS_DISTRO=kinetic DEBUG_BASH=true DOCKER_BASE_IMAGE=arm64v8/ros:kinetic-perception INJECT_QEMU=aarch64 DOCKER_PULL=false
```
`image: lsolankaflya/atlassian-default-image-qemu:19.04` is the `atlassian/default-image:2` that I rebuilt for 19.04.

It looks like the binfmt_misc is still not loaded in the kernel, since I still get the "exec format" error:

```
+ rm -rf /tmp/tmp.CaE8jyR2rU
Building image 'industrial-ci/kinetic/arm64v8/ros:kinetic-perception-qemu':
+ export DOCKER_IMAGE=industrial-ci/kinetic/arm64v8/ros:kinetic-perception-qemu
+ DOCKER_IMAGE=industrial-ci/kinetic/arm64v8/ros:kinetic-perception-qemu
+ echo 'Building image '\''industrial-ci/kinetic/arm64v8/ros:kinetic-perception-qemu'\'':'
+ local dockerfile
++ ici_generate_default_dockerfile
++ local keycmd
++ '[' -n '' ']'
++ keycmd='apt-key adv --keyserver '\''hkp://keyserver.ubuntu.com:80'\'' --recv-key '\''C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654'\'''
++ cat
+ dockerfile='FROM arm64v8/ros:kinetic-perception-qemu
ENV ROS_DISTRO kinetic
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8
RUN echo '\''debconf debconf/frontend select Noninteractive'\'' | debconf-set-selections
RUN apt-get update -qq && apt-get -qq install -y apt-utils gnupg2 wget ca-certificates lsb-release dirmngr python-pip python3-pip

<snip>

+ docker build -t industrial-ci/kinetic/arm64v8/ros:kinetic-perception-qemu -
Sending build context to Docker daemon   2.56kB
Step 1/9 : FROM arm64v8/ros:kinetic-perception-qemu
 ---> fc1fda1d452d
Step 2/9 : ENV ROS_DISTRO kinetic
 ---> Running in 02cefbebbf0b
Removing intermediate container 02cefbebbf0b
 ---> e97ed454c152
Step 3/9 : ENV LANG C.UTF-8
 ---> Running in 4ea7a1e736dc
Removing intermediate container 4ea7a1e736dc
 ---> 9d1149ba6b4b
Step 4/9 : ENV LC_ALL C.UTF-8
 ---> Running in 0187cf163f2f
Removing intermediate container 0187cf163f2f
 ---> 3a9434f47db7
Step 5/9 : RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
 ---> Running in 1eb57d2c4bc6
standard_init_linux.go:207: exec user process caused "exec format error"
The command '/bin/sh -c echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections' returned a non-zero code: 1
```

So I don't think that would be a viable solution, unless I misunderstood something.
But thanks for help anyway.

ipa-mdl, 2019-10-24:
Qemu should have been copied in the `<snip>` section, as part of preparing `arm64v8/ros:kinetic-perception-qemu`.
Did you already try it locally?

ipa-mdl, 2019-10-24:
According to `https://jira.atlassian.com/browse/BCLOUD-15`, Bitbucket does not allow setting up `binfmt_misc`.
The host running Docker has to register it somehow.

lsolanka-flya, 2019-10-24:
> Qemu should have been copied in the `<snip>` section, as part of preparing `arm64v8/ros:kinetic-perception-qemu`.
> Did you already try it locally?

Yes, the build of the docker container runs fine locally, and then the subsequent stages run fine as well (source code build, and tests; sorry can't post the full output here...)

> According to https://jira.atlassian.com/browse/BCLOUD-15317, Bitbucket does not allow setting up `binfmt_misc`.
> The host running Docker has to register it somehow.

Yep, aware of this ticket.

ipa-mdl, 2019-10-24:
Does the Bitbucket host have the directory `/proc/sys/fs/binfmt_misc/`?

PR #237 (2017-12-31): Inject QEMU for cross-platform build
If `DOCKER_IMAGE` or `DOCKER_BASE_IMAGE` point to a different platform, then `INJECT_QEMU` can be set to  the appropriate version that should get injected automatically.

For example for ARM builds, `INJECT_QEMU=arm` has to be set.

Please note:
* For this to work the `qemu-user-static` package needs to be installed on the host (or the CI machine): `sudo apt-get install qemu-user-static`
* There is no need to inject QEMU if it is already present in the selected image
* The build is much slower, it is not suitable for regular CI checks. It might be better to [... cut]