ros-industrial/industrial_ci #239: Accessing private Docker registry requires "docker login" early

130s, 2017-12-02:
I don't know whether this issue is general or not. Maybe `Gitlab` specific.

On `Gitlab` each repository works as a Docker repo as well (which is very nice IMO btw). By passing the name of the registered image in `DOCKER_IMAGE` I get:

```
++ docker pull registry.gitlab.com/ORG_XXX/REPO_DDD:DOCKERIMG_TAG
Error response from daemon: Get https://registry.gitlab.com/v2/ORG_XXX/REPO_DDD/manifests/DOCKERIMG_TAG: denied: access forbidden
```

I think it's because login required ([ref from gitlab.com](https://docs.gitlab.com/ce/ci/docker/using_docker_build.html#using-the-gitlab-container-registry):

```
     - docker login -u gitlab-ci-token -p $CI_JOB_TOKEN registry.example.com
```

If I'm right, then is there a way to run login in industrial_ci?

`DOCKER_IMAGE` is used in `ici_require_run_in_docker` that is run [at the very beginning of `source_tests.sh`](https://github.com/ros-industrial/industrial_ci/blob/adca7090fd7b8be22b0e0034e7668f7bf5ffc54b/industrial_ci/src/tests/source_tests.sh#L25), so I can't think of a way to run anything before that.

(Btw I'm not sure if this is the same for public repos.)

ipa-mdl, 2017-12-02:
At first I haven't read your desctiption properly and thought you were trying to use the images in a script.
Even this is not the case, I came up with #240 ;)

Your usecase should work out of the box.
I guess you have copied the wrong line; for accessing the internal repository it is:
 ```
docker login -u gitlab-ci-token -p $CI_JOB_TOKEN $CI_REGISTRY
```

130s, 2017-12-02:
Addressed via #240.

> for accessing the internal repository it is:
>
> ```
> docker login -u gitlab-ci-token -p $CI_JOB_TOKEN $CI_REGISTRY
> ```

That seems to be correct. Thank you!

closed 2017-12-02 by 130s

cscbai, 2021-09-02:
`docker login -u $CI_REGISTRY_USER -p $CI_JOB_TOKEN $CI_REGISTRY`

Same effect, but using variable for username