ros-industrial/industrial_ci #241: Using a Docker image already containing the target repo's master branch

130s, 2017-12-03:
I intend to use a Docker image where the source of the target repo is already cloned in, then it only needs to checkout the pull request's branch.
(I do this in order to save time as the target repo is gigabytes large (using Git LFS etc.) and CI often hits the CI's time limit.)

Currently this doesn't work as the workspace is read-only, so all attempts I've tried to update the target repo failed. I do not have a decisive opinion to an open PR https://github.com/ros-industrial/industrial_ci/pull/123

What will be the best way to do with industrial_ci?

130s, 2017-12-03:
A little more description for my usecase might be necessary.

A test in the target repo runs a downstream tests, and it requires a specific branch that's also available in a PR to the downstream repo. In other words, a PR to repo-A always comes with a PR to the downstream repo-B. To save the CI exec time, the aforementioned Docker image contains downstream pkgs as well.

I might have been wrong in my 1st post https://github.com/ros-industrial/industrial_ci/issues/241#issue-278761716 in that the source of the target repo should be even cloned to inside of Docker without problems. So my real problem might be to update downstream code in the pre-built Docker image during industrial_ci process.

Anyways in a nutshell, how to change the code in the workspace during industrial_ci process?

Hope this makes sense and also I'm not making an XY problem..

ipa-mdl, 2017-12-03:
> Anyways in a nutshell, how to change the code in the workspace during industrial_ci process?

rosdep?
If you need to overlay packages, you can even setup workspace chaining.

ipa-mdl, 2019-05-15:
#361 will just overwrite the code

closed 2019-08-18 by ipa-mdl