ros-industrial/industrial_ci #245: New release 201712

130s, 2017-12-10:
@ipa-mdl Since there are a bunch of good updates since the last release (i.e. just a tag and changelog), I like to tag and make an announcement (otherwise I'm afraid few users would notice these updates).
I'd rather prioritize making an announcement very soon, say Monday this week. I personally don't think any new action is needed before that.

For a PR for document update, while I ask you to review technical accuracy, I'm curious to ask for the community to take a look as well to see if the document makes sense in the first place from users' perspective.

Current version number is 0.4.0. Can the next version be 0.5.0?

ipa-mdl, 2017-12-11:
I am rather busy in the next days.
If you don't want to wait, just go ahead :+1: 

0.5.0 is reasonable :)

130s, 2017-12-13:
[Done](https://discourse.ros.org/t/industrial-ci-0-5-0-available-abi-compatibility-check-catkin-lint-arm-support-etc/3403).

closed 2017-12-13 by 130s