ros-industrial/industrial_ci #247: [Travis] Invalid SSH keys/config for wstool update

ipa-jba, 2017-12-15:
To fix the error ocurring since the latest update of travis I had to change the permissions of the .ssh folder.

See: https://github.com/ipa-jba/industrial_ci/commit/3cb5be401afac22d76388bd14f71719002f7d58e

@ipa-fxm

ipa-mdl, 2017-12-15:
I think this is unrelated to the new travis image, but might be an issue with the new docker version.
`docker cp` used to create files as root, but this seems to have changed.

This new version broke some other things (not related to `industrial_ci`) for me as well.

ipa-mdl, 2017-12-15:
Looks like a [bug](https://github.com/moby/moby/issues/34096) in docker.

ipa-mdl, 2017-12-29:
A fix for docker was [proposed](https://github.com/moby/moby/pull/35874).

ipa-mdl, 2019-05-15:
I guess this issue was resolved.

closed 2019-05-15 by ipa-mdl