ros-industrial/industrial_ci #249: Gitlab CI gets stuck during catkin run_tests

130s, 2017-12-17:
Similar but different from #204, I've been seeing a job gets stuck and hits the Gitlab CI's 1hr limit. I'm not yet sure if this is `industrial_ci`-specific issue or not.

At least it's been happening since the end of November. The particular line that job gets stuck varies, but it's always during build test (`catkin run_tests`).

Using the same Docker image locally on my computer, the same tests pass (running by the following). This tells me the issue is probably not in the Docker image I use:

```
$ lsb_release -a|grep -i Desc
No LSB modules are available.
Description:    Ubuntu 14.04.5 LTS
$ export CATKIN_PARALLEL_JOBS=-j2
$ export CATKIN_PARALLEL_TEST_JOBS=-j1
$ export ROS_PARALLEL_JOBS=-j2
$ export ROS_PARALLEL_TEST_JOBS=-j1
$ export PKGS_DOWNSTREAM=pmg
$ if [ "${TARGET_PKGS// }" == "" ]; then export TARGET_PKGS=`catkin_topological_order ${TARGET_REPO_PATH} --only-names`; fi
$ if [ "${PKGS_DOWNSTREAM// }" == "" ]; then export PKGS_DOWNSTREAM=$( [ "${BUILD_PKGS_WHITELIST// }" == "" ] && echo "$TARGET_PKGS" || echo "$BUILD_PKGS_WHITELIST"); fi
$ catkin build $OPT_VI --summarize  --no-status $BUILD_PKGS_WHITELIST $CATKIN_PARALLEL_JOBS --make-args $ROS_PARALLEL_JOBS            ; fi
$ catkin run_tests $OPT_VI --no-status $PKGS_DOWNSTREAM $CATKIN_PARALLEL_TEST_JOBS --make-args $ROS_PARALLEL_TEST_JOBS --
```

Related or not, I noticed in a lot of cases the job remains at the following line for awhile. But eventually moves forward.
```
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function catkin_build returned with code '0' after 11 min 50 sec
:
++ /usr/bin/catkin run_tests -w /root/catkin_ws --no-status pmg -p1 --make-args -j1 --
==> Expanding alias 'run_tests' from 'catkin run_tests -w /root/catkin_ws --no-status pmg -p1 --make-args -j1 --' to 'catkin build --verbose --catkin-make-args run_tests -- -w /root/catkin_ws --no-status pmg -p1 --make-args -j1 --'

Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /root/catkin_ws

Source Space:       [exists] /root/catkin_ws/src
Log Space:          [exists] /root/catkin_ws/logs
Build Space:        [exists] /root/catkin_ws/build
Devel Space:        [exists] /root/catkin_ws/devel
Install Space:      [exists] /root/catkin_ws/install
DESTDIR:            [unused] None

Devel Space Layout:          linked
Install Space Layout:        merged

Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        -j1
Additional catkin Make Args: run_tests
Internal Make Job Server:    True
Cache Job Environments:      False

Whitelisted Packages:        None
Blacklisted Packages:        None

Workspace configuration appears valid.

[build] Found '15' packages in 0.0 seconds.
:
Subprocess  > pmg:check `cd /root/catkin_ws/build/pmg; catkin build --get-env pmg | catkin env -si  /usr/bin/make cmake_check_build_system; cd -`
Output     << pmg:check /root/catkin_ws/logs/pmg/build.check.000.log
/usr/bin/cmake -H/root/catkin_ws/src/metheny_test/pmg -B/root/catkin_ws/build/pmg --check-build-system CMakeFiles/Makefile.cmake 0

cd /root/catkin_ws/build/pmg; catkin build --get-env pmg | catkin env -si  /usr/bin/make cmake_check_build_system; cd -

Finished   << pmg:check
Starting   >> pmg:make
Subprocess  > pmg:make `cd /root/catkin_ws/build/pmg; catkin build --get-env pmg | catkin env -si  /usr/bin/make run_tests --jobserver-fds=6,7 -j; cd -`
```

ipa-mdl, 2017-12-18:
I am not convinced that the `*_PARALLEL*_JOBS` settings actually work as expected.
Depending on the docker host, some part of the test might die silently and block infinitely.

130s, 2018-01-21:
I kind of concluded that there might be something in the way Gitlab CI handles large size of Docker container. This issue typically happens to me using a Docker image of 10+GB, even though the same `catkin run_tests` step finishes without any issue locally. At least this has nothing to do with industrial_ci.

miguelprada, 2018-03-28:
I've recently started having similar issues as well, and so far it has only happened to me on indigo/trusty-based tests. Quite frequently, they get stuck on seemingly random lines, always during the `catkin_run_tests` step.

@130s have you found anything else about what the cause might be?

130s, 2018-03-28:
I haven't found anything deterministic. I think this is very unlikely industrial_ci's issue though. Only from experience I can tell:
- We run a few hundreds of test cases, each of which load 10+ MB data as an image topic and image processing runs. Gitlab CI gets stuck way more often when we run full set of test cases, but stuck way less under 100 cases.
- We've only run those tests on ROS Indigo (we're not ready to try on Kinetic). So kind of the same situation with you for this regard.
- Aside from stuck issue but just as an improvement, I found `rosdep install` takes very long to install basically all necessary ROS packages and system dependencies (I use [ros-industrial/docker/ci](https://github.com/ros-industrial/docker/tree/master/ci), which comes with nearly no ROS pkg pre-installed). This holds even when I run `rosdep install` locally on the same Docker image parent. So I ended up making an "intermediate" image where rosdep is already run.

I've also asked Gitlab support to look into this but I haven't got much help other than telling me to extend the CI time limit to more than 60mins (which is not a direct fix but helped to amend the situation, although some jobs sometimes still exceed even 3hrs due to stalling).

miguelprada, 2018-03-28:
> - We run a few hundreds of test cases, each of which load 10+ MB data as an image topic and image processing runs. Gitlab CI gets stuck way more often when we run full set of test cases, but stuck way less under 100 cases.

In my case I don't run nearly as many test cases. It's just 3 cases with a few tests in each.

> - Aside from stuck issue but just as an improvement, I found rosdep install takes very long to install basically all necessary ROS packages and system dependencies (I use ros-industrial/docker/ci, [...]

I see this too. Weird thing is that same job in kinetic/xenial installs the dependencies **much** faster.

> I've also asked Gitlab support to look into this but I haven't got much help other than telling me to extend the CI time limit to more than 60mins (which is not a direct fix but helped to amend the situation, although some jobs sometimes still exceed even 3hrs due to stalling).

The stalling jobs seem to be stuck forever in my case, since they usually take around 20 minutes. Except when they don't :sweat_smile:

I'll try increasing the time limit to see if they eventually get un-stuck, and play with other variables to see if I figure out anything, but its kind of painful to iterate on this since its very intermittent and every iteration takes the aforementioned 20 minutes.

ipa-mdl, 2018-03-28:
Can you see what part exactly got stuck (e.g. `docker exec`)?
Are you running Docker on `aufs` or `overlay`?
Depending on the config file access (rosdep installs..) might be slow and sometime file access fails for some reason, but does not get reported..

miguelprada, 2018-04-03:
> Can you see what part exactly got stuck (e.g. docker exec)?

Not sure what you mean. As I said above, when jobs get stuck, it always happens during the `catkin_run_tests` step.

> Are you running Docker on `aufs` or `overlay`?

According to `docker info`, I'm running on `overlay2`.

ipa-mdl, 2018-04-03:
> Not sure what you mean. As I said above, when jobs get stuck, it always happens during the catkin_run_tests step.

I meant to login to the running docker container (if possible) and analyze the processes, e.g. with pstree, htop, etc.
Sometimes jobs freeze on IO accesss with overlay problems. Then the access problems should get reported in syslog on the host.

miguelprada, 2018-04-03:
> I meant to login to the running docker container (if possible) and analyze the processes, e.g. with pstree, htop, etc.
> Sometimes jobs freeze on IO accesss with overlay problems. Then the access [...]

:+1: Will have a look next time the job freezes and report back.

130s, 2018-05-30:
Most likely unrelated, as this ticket is about `run_tests` step, but https://github.com/ros-infrastructure/ros_buildfarm/issues/535 is interesting to me as I've seen `apt-get` and/or `rosdep install` within industrial_ci often take very long.

ipa-mdl, 2018-05-30:
@130s: It's unrelated, but might be a reason for slow installs. In addition unnecessary dependencies migth got install. This can be fine-tuned with `ROSDEP_SKIP_KEYS`.

ipa-mdl, 2018-06-03:
https://github.com/catkin/catkin_tools/issues/379 might be related.

130s, 2018-07-11:
Not sure if this is even helpful, but I've stopped seeing this issue (@geoffreychiou may agree) after we reduced the size of the repositories cloned into the Docker image (from around 10GB to 2 or 3 GB).

130s, 2018-07-31:
I meant the size of checked out resources, excluding build artifact. In our case the majority of resource was image files.

Combined with your experience @v-lopez, I'm more inclined to say there's something about the size of data loaded into RAM?

v-lopez, 2018-07-31:
Well this happens on a physical computer with 32GB of RAM, and it happens long after building all our code, which is more memory intensive.

It occurs when it is running the tests individually, at the end of the execution of one of the tests it just gets stuck, but the test is fully ended and it's on ROS/Catkin Tools code space when this happens.

I'm quite sure it's related to https://github.com/catkin/catkin_tools/issues/379, I added a callstack there.
Basically because to us it occurs in a docker container launched from jenkins, we're not using Gitlab CI for this.

Below is the log, it got stuck for 9 hours after the last line. If I kill the catkin build process via terminal, the build resumes without issues.

>  [rostest][INFO] 2018-07-30 23:55:49,562: test [tasks_container_priority_gtest] finished
[rostest][INFO] 2018-07-30 23:55:49,563: test [tasks_container_priority_gtest] results are in [/tmp/test_results/repos/pal_wbc_controller/rosunit-tasks_container_priority_gtest.xml]
[rostest][INFO] 2018-07-30 23:55:49,565: test [tasks_container_priority_gtest] results summary: 0 errors, 0 failures, 2 tests
[rostest][INFO] 2018-07-30 23:55:49,566: [ROSTEST] test [tasks_container_priority_gtest] done
[rostest][INFO] 2018-07-30 23:55:49,566: tearDown[/tmp/test_repositories/src_repository/pal_wbc/pal_wbc_controller/ggtest/tasks_container_priority.test]
[roslaunch][INFO] 2018-07-30 23:55:49,566: runner.stop()
[rostest][INFO] 2018-07-30 23:55:49,567: shutting down processing monitor...
[roslaunch][INFO] 2018-07-30 23:55:49,567: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, started daemon 140686182921984)>
[roslaunch.pmon][INFO] 2018-07-30 23:55:49,567: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, started daemon 140686182921984)>
[roslaunch.pmon][INFO] 2018-07-30 23:55:49,666: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140686182921984)>
[roslaunch.pmon][INFO] 2018-07-30 23:55:49,666: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140686182921984)>: remaining procs are []
[roslaunch.pmon][INFO] 2018-07-30 23:55:49,671: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2018-07-30 23:55:49,671: ProcessMonitor exit: pmon has shutdown
[rostest][INFO] 2018-07-30 23:55:49,672: ... shutting down processing monitor complete
[roslaunch.pmon][INFO] 2018-07-30 23:55:49,672: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 140686182921984)>
[rostest][INFO] 2018-07-30 23:55:49,673: rostest teardown /tmp/test_repositories/src_repository/pal_wbc/pal_wbc_controller/ggtest/tasks_container_priority.test complete
[rostest][INFO] 2018-07-30 23:55:49,705: finally rostest parent tearDown [<rostest.rostest_parent.ROSTestLaunchParent object at 0x7ff40e9f1990>]
[roslaunch][INFO] 2018-07-30 23:55:49,705: runner.stop()
[roslaunch][INFO] 2018-07-30 23:55:49,706: ... roslaunch runner has already been stopped
[rostest][INFO] 2018-07-30 23:55:49,706: calling pmon_shutdown
[rostest][INFO] 2018-07-30 23:55:49,706: ... done calling pmon_shutdown
[rospy.core][INFO] 2018-07-30 23:55:49,706: signal_shutdown [atexit]

ipa-mdl, 2019-07-14:
Are you still facing this issue?

ipa-mdl, 2019-08-18:
I will close this now, because it seems to be a problem with catkin-tools.
Other builders are now supported as well.

closed 2019-08-18 by ipa-mdl