ros-industrial/industrial_ci #252: Remove testing installed test files?

ipa-mdl, 2017-12-18:
By default `industrial_ci` will try to run all installed test files.
I just discovered that the retry feature is acutally broken because eventually the test will fail even if the retry was successful.

This made me wonder: Does someone use this feature?
The way catkin-tools organizes the (install) workspace should have almost the same effect..

130s, 2019-06-25:
I looked around my internal corporate repos and there are a few cases where `.test` files are installed and running those files from installed space. That said, for us particularly dropping this feature wouldn't matter as long as tests can be run in other ways.

Which retry feature are you talking about? `test` node in `.test` file?

ipa-mdl, 2019-06-25:
> That said, for us particularly dropping this feature wouldn't matter as long as tests can be run in other ways.

These tests can still be executed from CMakeLists.txt.
With `colcon` they will get run from the install space as well.
(I can't say for sure for `catkin_tools`..)
 
> Which retry feature are you talking about? test node in .test file?

https://github.com/ros-industrial/industrial_ci/blob/legacy/industrial_ci/src/tests/source_tests.sh#L269
But it might have been fixed in the meantime

ipa-mdl, 2019-08-18:
removed in #361

closed 2019-08-18 by ipa-mdl