ros-industrial/industrial_ci #253: `catkin_test_results --verbose` explodes travis log beyond max

ipa-fxm, 2018-01-03:
with the changes introduced in #251 (related to #118), our Travis jobs fails due to exceeding log length maximum
(we haven't seen this earlier due to other Travis/Docker issues - see #247 #248)

```
The log length has exceeded the limit of 4 MB (this usually means that the test suite is raising the same exception over and over).

The job has been terminated
```

I can ensure that "the test suite is raising the same exception over and over" is **not** true in our case!

To give some numbers:
 - all CI steps from start to build-step produce about 2000 lines of output
 - the test-step (--verbose) produces more than 8000 lines of output....

Is there a way to fix #118 without using `--verbose` during the test-step?

Our `travis.yml` looks pretty standard (and has worked before):
```
sudo: required
dist: trusty
language: generic
notifications:
  email:
    on_success: change
    on_failure: always
env:
  global:
    - ROS_REPO=ros
    - NOT_TEST_INSTALL=true
  matrix:
    - ROS_DISTRO=indigo UPSTREAM_WORKSPACE=file
    - ROS_DISTRO=indigo UPSTREAM_WORKSPACE=file $ROSINSTALL_FILENAME="custom.rosinstall"
    - ROS_DISTRO=kinetic UPSTREAM_WORKSPACE=file
matrix:
  allow_failures:
    - env: ROS_DISTRO=kinetic UPSTREAM_WORKSPACE=file
install:
  - git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
script:
  - .ci_config/travis.sh
#  - ./travis.sh  # Enable this when you have a package-local script 

```

ipa-fxm, 2018-01-03:
Could [this](http://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html?highlight=catkin_run_tests#building-and-running-tests) be used to get the proper non-zero return code for failing tests, i.e. calling `catkin_test_results` with the respective `build/PKG_NAME` argument - instead of the `--verbose` argument without the folder argument?

 > For catkin packages and the run_tests target, failing tests will not result in an non-zero exit code. So if you want to check for failing tests, use the catkin_test_results command like this:
 >
 > $ catkin_test_results build/<package name>
 > 
 > The result code will be non-zero unless all tests passed.

ipa-fxm, 2018-01-03:
I'm currently testing https://github.com/ipa-fxm/industrial_ci/commit/f666f6b3e003a82a9504949bd97deeefd99a5a61 in our repo
I'll provide a PR if it helps...

ipa-fxm, 2018-01-03:
It rather is [this line](https://github.com/ros-industrial/industrial_ci/blob/dad31f8cec35a29e3b7d75cda3fc8bfcf46623f9/industrial_ci/src/tests/source_tests.sh#L175) and not [this line](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/tests/source_tests.sh#L185)

```
Expanding alias 'run_tests' from 'catkin run_tests -w /root/catkin_ws --no-status PACKAGE_LIST -p4 --make-args -j8 --' to 'catkin build --verbose --catkin-make-args run_tests -- -w /root/catkin_ws --no-status PACKAGE_LIST -p4 --make-args -j8 --'
```

Any idea what can be done about this?
I haven't set `$OPT_VI` via [`VERBOSE_OUTPU`](https://github.com/ros-industrial/industrial_ci/blob/dad31f8cec35a29e3b7d75cda3fc8bfcf46623f9/industrial_ci/src/tests/source_tests.sh#L28-L32) - which according to https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#optional-environment-variables is "not set" by default

ipa-mdl, 2018-01-04:
I don't see any reason for the verbose flag to influence your builds.
Its effect is "Show all tests which have errors or failed" (from `catkin_test_results --help`).
So if no test has failed, it will just print the one-line summary. If tests fail, you will get a proper output.
For successful jobs, even less lines shoud get printed..

Your builds might be affected by https://github.com/ros-industrial/industrial_ci/pull/244.
Can you provide me the logs (e.g. private gist)?

ipa-mdl, 2018-01-04:
Ah, now I have seen which verbose flag you are talking about (yep, side scrolling helped..).
The expanded `run_tests` command is:
```
catkin build --verbose --catkin-make-args run_tests -- -w /root/catkin_ws --no-status PACKAGE_LIST -p4 --make-args -j8 --
```
This flag has nothing to do with #251.

ipa-fxm, 2018-01-04:
@ipa-mdl  
yes the `--verbose` in `catkin_test_results` is ok...I re erted this already
the problem is the `--verbose` which gets added under the hood when expanding the alias `run_tests`

ipa-mdl, 2018-01-04:
I guess the verbose flag is needed to see the output of rostest etc.
As a compromise the tests (and their dependencies) could be built beforehand (`tests` make target).

However, if your build is affected that much by #244 then you should definitely review the dependencies.
I am not sure which dependencies are taken into account by catkin-tools.

ipa-fxm, 2018-01-04:
PR is provided in #255 

explanation:
https://github.com/catkin/catkin_tools/blob/master/catkin_tools/config.py#L39 defines the alias `run_tests` for `catkin_tools` to `build --verbose --catkin-make-args run_tests --''
Unfortunately, this blows up the travis log beyond the allowed maximum of 4MB (>> 10000 lines)
In #255, I perform the suggested expansion of the alias `run_tests` but removed the explicit `--verbose` option
`--verbose` can still be configured using `$OPT_VI` via `VERBOSE_OUTPUT` parameter in the `.travis.yml`

In contrary to @ipa-mdl's assumption, about needing the `--verbose` in `run_tests` to see the output of `rostest`, all information needed to debug failing tests is still displayed during the `catkin_test_results --verbose $CATKIN_WORKSPACE || error` step

ipa-mdl, 2018-01-04:
>  all information needed to debug failing tests is still displayed during the "catkin_test_results --verbose $CATKIN_WORKSPACE || error"-step

It not just about debugging..
Some users might want to see the output of rostest (or any other test suite).
Your problem is not the output of the tests, but the output of the compilation of other packages.
(please compare the number of build packages in `run_tests` before and after #244)

 I would like to have these things split up.
Only the build step shoud be affected by `VERBOSE_OUTPUT`.

ipa-fxm, 2018-01-04:
Ok, I'm investigating the effects of "reverting #244" and will get back tomorrow...

ipa-fxm, 2018-01-04:
Good things first:
Both variants V1 (expand `run_tests` without `--verbose`) and V2 (re-add `--no-deps`) work for our use-case, i.e. travis logs stay below 4MB

As requested:
 - V1 (expand `run_tests` without `--verbose`):
   ```
   [build] Summary: All 86 packages succeeded!
   [build]   Ignored:   71 packages were skipped or are blacklisted.
   [build]   Warnings:  8 packages succeeded with warnings.
   [build]   Abandoned: None.
   [build]   Failed:    None.
   [build] Runtime: 6 minutes and 52.6 seconds total.
   Summary: 263 tests, 0 errors, 0 failures, 0 skipped
   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
   Function catkin_run_tests returned with code '0' after 7 min 0 sec 
   ```
 - V2 (re-add `--no-deps`)
   ```
   [build] Summary: All 26 packages succeeded!
   [build]   Ignored:   131 packages were skipped or are blacklisted.
   [build]   Warnings:  2 packages succeeded with warnings.
   [build]   Abandoned: None.
   [build]   Failed:    None.
   [build] Runtime: 1 minute and 37.9 seconds total.
   Summary: 109 tests, 0 errors, 0 failures, 0 skipped
   <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
   Function catkin_run_tests returned with code '0' after 1 min 45 sec 
   ```

Conclusion:
We could life with one and/or the other variant, but currently none of them is configurable, isn't it?

@ipa-mdl 
Would you be ok merging a PR for one (#255) and/or the other (revert #244) solutions?
Or do you have another approach in mind? I can help implement/test...

ipa-mdl, 2018-01-05:
I don't want to revert #244 because it fixes a dormant issue with unbuilt dependencies.
(Setting `--no-deps` only might not test against all the packages you want it to..)
And I don't want to merge #255 (as is) because it would couple the build verbosity with the run_test verbosity.

It looks like `run_tests` without `--no-deps` is running the tests of the dependencies as well, which is not what we want.
However, we need a way to build all dependencies before runnng the tests.

Something like:
* build all dependencies, verbose if `VERBOSE_OUTPUT`
* build all selected tests (with --no-deps*), verbose if `VERBOSE_OUTPUT`
* run all selected tests (with --no-deps*), verbose

(* --no-deps should only be set if `PKGS_DOWNSTREAM` is not empty)

PR #256 (2018-01-14): Fix run_tests verbosity
This is my attempt to fix #253, as a replacement for #255 

It reverts #244, but sets `TARGET_PKGS` to all packages to address #232.
To make sure that all downstream packages and their dependencies get built, a dedicated step was added.

In addition `tests` will be built before `run_tests` to build the tests with the build options. 

@miguelprada: please test your bare rosinstall usecase
@ipa-fxm: please test with your setup
closed 2018-01-14 by 130s