ros-industrial/industrial_ci #254: travis failing

jihoonl, 2018-01-04:
I am experiencing industrial_ci travis job failure in [web_video_server](https://github.com/RobotWebTools/web_video_server). Do you have any insight why it is failing?

* [Configuration](https://github.com/RobotWebTools/web_video_server/blob/develop/.travis.yml)
* [Log](https://travis-ci.org/RobotWebTools/web_video_server/jobs/323669365)

130s, 2018-01-04:
(Updated title to describe the issue better)

The direct cause of failure is that [`docker` is not available](https://travis-ci.org/RobotWebTools/web_video_server/jobs/323669365#L174). Looks like the job passed [a month ago](https://travis-ci.org/RobotWebTools/web_video_server/jobs/304806570).

I don't know why, but [apt-get failed to access many sites for retrieving updates](https://travis-ci.org/RobotWebTools/web_video_server/jobs/323669365#L105). As an easy-slacker attempt, I'm curious to see if restarting the same job fixes this or not.

jihoonl, 2018-01-04:
not sure.. I tried multiple times and confirmed that the problem persists.

ipa-mdl, 2018-01-04:
Duplicate of #167, your config is not valid..
Please set `dist: trusty`.

jihoonl, 2018-01-08:
Ok. thanks it works.

ipa-mdl, 2018-01-31:
This seems to confuse a number of users.

ipa-mdl, 2019-05-15:
> Please set dist: trusty.

This can be removed now.
Travis is already using `trusty` per default and will soon migrate to `xenial`.

closed 2019-05-15 by ipa-mdl