ros-industrial/industrial_ci #258: ABI check feature not functional on ROS Indigo?

130s, 2018-01-10:
Reported in https://github.com/ros-drivers/openni2_camera/pull/75#issuecomment-351324425

ipa-mdl, 2018-01-14:
It works with actionlib https://travis-ci.org/ros-industrial/industrial_ci/jobs/328743544

ipa-mdl, 2018-01-14:
I think the issue here that nodelet libraries (or plugins in general) do not have an API/ABI and therefore abi-complicance-checker fails.
On lunar some symbols from OpenNI get exported, so this is just coincident..

We have some options:
1. implement blacklist, e.g. `ABI_CHECK_BLACKLIST='openni2_camera_nodelet'`
2. skip libraries with this error code
3. skip all plugins

The first option is very specific, less error-prone, not user-friendly (user can and must speficy it).
The second option might blacklist libraries that don't contain any symbols by accident.
The third option ist hard to implement and might skip hybrid libraries that should get checked.

PR #369 (2019-05-22): Backport some colcon features
Partial backport of #361 

* align the code/behavior between both version
* shellcheck :)
*  fix #258, fix #322 
* backport of CMAKE_ARGS (highly perferred over `CATKIN_CONFIG`)

In addition it should simplify the review.

# CoS (Condition of Satisfaction, suggested by @130s)
- Code review.
  - [x] @miguelprada 
  - [x] @130s
- (Locally) verify the change.
  - [x] @130s ABI error https://github.com/ros-industrial/industrial_ci/issues/258 -> Ongoing in https://github.com/130s/openni2_camera/pull/2
closed 2019-05-22 by 130s