ros-industrial/industrial_ci #260: apt source list won't be updated with a specified Docker image

130s, 2018-01-11:
When no Docker image was passed, `apt` source list [is set](https://github.com/ros-industrial/industrial_ci/blob/2c5ec54e7cdbc4aafe5be23b1aba4350800b5f77/industrial_ci/src/docker.sh#L223) but when an image specified, no apt source list addition/update occurs and the job fails because ROS related packages won't be found.

Do we want to improve this situation (e.g. check if proper source list is already set or not, and if not add a source list), or is there any issue with that?

130s, 2018-01-11:
I think it is already documented, so is a requirement currently.

```
DOCKER_IMAGE (default: not set): Selects a Docker images different from default one. Please note, this disables the handling of ROS_REPOSITORY_PATH and ROS_DISTRO as ROS needs already to be installed in the image.
```

closed 2018-01-11 by 130s

ipa-mdl, 2018-01-11:
You could set `DOCKER_BASE_IMAGE` instead, but it might mix different versions..