ros-industrial/industrial_ci #264: Copy source of target repo with the org name?

130s, 2018-01-14:
`industrial_ci` [symlinks the target repo from `catkin_ws/src`](https://github.com/ros-industrial/industrial_ci/blob/2c5ec54e7cdbc4aafe5be23b1aba4350800b5f77/industrial_ci/src/tests/source_tests.sh#L118).

Meanwhile, `wstool` creates `.rosinstall` with `%ORG_NAME%/%REPO_NAME%` by default by `wstool scrape`. 

These 2 don't align well, when using a custom Docker image where there's already a repository of the target is but with `org/repo` structure, as later in CI process duplicate repo error occurs.

Respecting the popularity of `wstool`, it might make more sense to me in industrial_ci to create a symlink with org/repo structure.

Thoughts?

ipa-mdl, 2018-01-14:
I prefer the repository name only because it's the way `git clone` behaves.
I have now tried `wstool scrape` (for the first time), it does not enforce `org/repo` semantics, but just uses the path releative to the source space.
So this is just your user preference for the source layout.

That beeing said, we could introduce something like `TARGET_LOCAL_PATH`.
However, it is problematic if your image already contains the target repo, this needs extra handling.

IMHO this could be resolved cleanly if we would start using proper overlay and underlay workspaces.

ipa-mdl, 2018-01-14:
> when using a custom Docker image where there's already a repository of the target is but with org/repo

For now I would suggest to remove the duplicated repository in the `BEFORE_SCRIPT`.

130s, 2018-02-02:
> IMHO this could be resolved cleanly if we would start using proper overlay and underlay workspaces.

Do you mean industrial_ci handling over/underlay automatically? That might be interesting. Is it something like, all repos fetched by rosintall are laid under target repo?

ipa-mdl, 2019-05-15:
> IMHO this could be resolved cleanly if we would start using proper overlay and underlay workspaces.

Implemented in #361

closed 2019-08-18 by ipa-mdl