ros-industrial/industrial_ci #269: ABI check feature stuck on ROS Kinetic

130s, 2018-02-16:
On RViz repo https://github.com/ros-visualization/rviz/pull/1198#issuecomment-366115970 the Travis hit 2 hr (not even an hour) limit. I'm not even sure what is the critical error message.

Not sure if this is related to #258.

FYI @dhood the error looks odd to me so we'll discuss here, but one question. Have you run the job multiple times and seen the same? If you haven't can you try restarting?

ipa-mdl, 2018-02-16:
I don't think that this is an issue with `industrial_ci`.
It is not related to #258.

dhood, 2018-02-16:
I usually use jenkins and I haven't used travis much, so please keep that in mind in case there's something obvious I might be overlooking.

The four travis jobs for that PR have all ended with `The job exceeded the maximum time limit for jobs, and has been terminated.`. The 2hr build that you saw was just a sporadic thing, the other three timed out at 49min: @ipa-mdl pointed out there were some travis issues during that time that might have affected the reporting. Here are the builds:

https://travis-ci.org/ros-visualization/rviz/builds/342031995
https://travis-ci.org/ros-visualization/rviz/builds/342040557
https://travis-ci.org/ros-visualization/rviz/builds/342087177
https://travis-ci.org/ros-visualization/rviz/builds/342483044

They were at the last step when they got terminated, but I can't tell if they hang at that step, or if it just pushes them over the time limit. Locally it only takes 30min to run `industrial_ci/scripts/run_travis -` but the travis machines might have additional setup steps that take extra time?

ipa-mdl, 2020-04-08:
I will close this now, because I don't think that we can solve this on our side.

closed 2020-04-08 by ipa-mdl