ros-industrial/industrial_ci #272: Install additional package before 'git clone'

ipa-fxm, 2018-02-21:
For one of our repos, I need `git lfs` to be set up before downloading the repos in order to retrieve some files tracked under `git lfs` required for a successful `catkin  build`...

It's the following commands I need to perform (https://github.com/git-lfs/git-lfs/wiki/Installation)
```
curl -s https://packagecloud.io/install/repositories/github/git-lfs/script.deb.sh | sudo bash
sudo apt-get install git-lfs
git lfs install
```

How can I achieve this?
The `BEFORE_SCRIPT` is too late, isn't it? `BEFORE_SCRIPT` is executed before `catkin build` but after `git clone`, correct?

ipa-fxm, 2018-02-21:
Could I use the `DOCKER_RUN_OPTS` together with
```
RUN build_deps="curl ca-certificates" && \
    apt-get update && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends ${build_deps} && \
    curl -s https://packagecloud.io/install/repositories/github/git-lfs/script.deb.sh | bash && \
    DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends git-lfs && \
    git lfs install && \
    DEBIAN_FRONTEND=noninteractive apt-get purge -y --auto-remove ${build_deps} && \
    rm -r /var/lib/apt/lists/*
```

ipa-mdl, 2018-02-21:
Do you mean "git clone" or "wstool update"?
The first will be executed by Travis. The latter is run by `industrial_ci`.
Unfortunately, there is now way to to add a script before this step via config options.

You could build you custom base image to include git-lfs.

As a generic solution we could add a new hook `SETUP_SCRIPT` (?) that is run before anything else.

ipa-fxm, 2018-02-21:
I'm playing around on branch https://github.com/ipa-fxm/industrial_ci/tree/pass_custom_docker_recipes

but this still does not solve my problem for two reasons:
 - once the `generate dockerfile`-part is executed - of course - my repo is already cloned... :disappointed: 
 - I get the following error with my custom recipe
    ```
    W: Failed to fetch https://packagecloud.io/github/git-lfs/ubuntu/dists/trusty/main/source/Sources  
    W: Failed to fetch https://packagecloud.io/github/git-lfs/ubuntu/dists/trusty/main/binary-amd64/Packages  
    E: Some index files failed to download. They have been ignored, or old ones used instead.
    ```

I'd like to avoid maintaining a custom docker-file...as I'd like to benefit from updates being done from this repo :wink:

How would I implement such a `SETUP_SCRIPT`? Can you point me to a line where I would insert calling it?

ipa-fxm, 2018-02-21:
Actually, for my particular use-case atm, **before** `wstool update` would be sufficient...for now at least...the lfs-files I need for compiling my repo are in a repo added via rosinstall...

ipa-mdl, 2018-02-21:
> I'd like to avoid maintaining a custom docker-file...as I'd like to benefit from updates being done from this repo :winnk:

Thats why you would supply a baseimage, which extends ubuntu:xenial

> Can you point me to a line where I would insert calling it?

I have not found a proper location.
Appending so the docker script might be an option though.

ipa-mdl, 2018-02-21:
> Appending so the docker script might be an option though.

This might be an escping nightmare..
A `SETUP_SCRIPT` could be added here https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/docker.sh#L48

130s, 2018-12-03:
I sometimes wonder if this is possible as well.

Without much thought, I'm just experimenting to move `BEFORE_SCRIPT` to earlier phase.
- [code](https://github.com/130s/industrial_ci/commit/8d453428053d66446c0e0ed83075aaab782e5dd7) (or [branch](https://github.com/130s/industrial_ci/tree/mv_beforescript))
- [industrial_ci's CI run using my fork](https://travis-ci.org/130s/industrial_ci/builds/463013489)

ipa-mdl, 2018-12-03:
> I'm just experimenting to move BEFORE_SCRIPT to earlier phase.

This changes the behavior.
At the moment the `BEFORE_SCRIPT` can be used to update/blacklist some of the upstream repositories.

130s, 2018-12-04:
>>    I'm just experimenting to move BEFORE_SCRIPT to earlier phase.
> 
> This changes the behavior.

Confirmed. [One of the CI jobs captured the regression](https://travis-ci.org/130s/industrial_ci/jobs/463013552#L474). Cool.

I think there is a need for injecting a set of custom commands execution to happen earlier than the current `BEFORE_SCRIPT` location. Using custom Docker image indeed works around, but it's not always possible. Having industrial_ci supporting that usecase would be great.

(Although I can't recall why BEFORE_SCRIPT ended up being at the current location, it could have done earlier than it does now, couldn't it? But as discussed above we can't move it.)

Should we add a (yet another) variable for that? I'm down for this rather than restricting usecases.

ipa-mdl, 2019-08-18:
The new hook system (#361) solves this.

closed 2019-08-18 by ipa-mdl