ros-industrial/industrial_ci #273: How to fix launch test depend error ?

RyodoTanaka, 2018-03-06:
[On our project](http://github.com/Nishida-Lab/motoman_project), we want to do launch test.
But we have dependencies between our original packages.
That's why, when we want to do launch test using `catkin tests` command, it returns error on Travis.
Here is the detail.

# Problem
Because of dependencies between our original packages, the `catkin test` returns error.
The result is shown in following...
https://travis-ci.org/Nishida-Lab/motoman_project/builds/334956162

# What we did (tried to fix it)
Export depend package names into `TARGET_PKGS` environment argument.
But, it did not work.

I have no idea what should I do...
Thank you forward.
Best Regards.

ipa-mdl, 2018-03-06:
According to your travis log, you are using a [patched version](https://github.com/ros-industrial/industrial_ci/compare/master...MoriKen254:add-source) of `industrial_ci`

What error did you get without your patches?
You should not alter `ROS_PACKAGE_PATH` in any way!
In addtion the devel space won't get build by default with `industrial_ci`.

> Because of dependencies between our original packages

This is true for almost all repositories and does not lead to problems (normally).

RyodoTanaka, 2018-03-07:
@ipa-mdl 
> What error did you get without your patches?
Here is the result without our patches.
https://travis-ci.org/Nishida-Lab/motoman_project/jobs/342213668

> You should not alter ROS_PACKAGE_PATH in any way!
Because of the error, we wanted tried to set and see that.
But, I understand that we should not alter 'ROS_PACKAGE_PATH`.

ipa-mdl, 2018-03-07:
Your job fails at the compile step:
```
/root/catkin_ws/src/motoman_project/motoman_sia5_moveit_plugins/src/motoman_sia5_arm_ikfast_solver.cpp:7868:71: error: ‘>>’ should be ‘> >’ within a nested template argument list
                                                                 IkReal>>
```
C++03 interprets ">>" as a bit shift.

RyodoTanaka, 2018-03-07:
@ipa-bnm 
Sorry for late reply.
> Your job fails at the compile step:

Actually, it was that.
I'm sorry to bother you that the error was not related with my question.
Any way, I fixed it, and encountered other 2 errors.

### Error 1 : It not start to build
Here is the Travis error result.
https://travis-ci.org/Nishida-Lab/motoman_project/builds/350341210
I do not understand why it is occurred.

### Error 2 : Dependency Error
On [the other our package](https://github.com/CIR-KIT/fourth_robot_pkg/tree/indigo-devel), I tested roslaunch test.
On that package, we have dependencies between original package.
And we got following error.
https://travis-ci.org/CIR-KIT/fourth_robot_pkg/builds/350384691
It says they could not find the path for original package, I think.
But I do not know how to fix that....

ipa-mdl, 2018-03-07:
> Error 1 : It not start to build
> I do not understand why it is occurred.

Please restart the job.
The build did start, but the output might got suppressed.
You could try to set `VERBOSE_OUTPUT=true`
However, I would recommend to run in locally first (`run_travis` or `run_ci`).

> Error 2 : Dependency Error

Just see the logs:
> [/root/catkin_ws/src/fourth_robot_pkg/fourth_robot_gazebo/launch/husky_playworld.launch]:
	Invalid <arg> tag: fourth_robot_description

This just tells you that something is wrong with your launch files.
That's exactly what these tests are for.

RyodoTanaka, 2018-03-07:
@ipa-mdl 
### Error 1
I will try it and report later.

### Error 2
> This just tells you that something is wrong with your launch files.
> That's exactly what these tests are for.

I launched it and there is no error on my local environment.
And also, where I use `fourth_robot_description` is following.
https://github.com/CIR-KIT/fourth_robot_pkg/blob/indigo-devel/fourth_robot_gazebo/launch/husky_playworld.launch#L4

I do not understand which part of my code is wrong....

RyodoTanaka, 2018-03-07:
@ipa-mdl 
I tried to set 'VERBOSE_OUTPUT=true` but, it does not works....
Here is the result on Travis.
https://travis-ci.org/Nishida-Lab/motoman_project/builds/350427009
> However, I would recommend to run in locally first (run_travis or run_ci).

Honestly, I do not have an environment about docker on my local PC....
So, I will set my environment.

ipa-mdl, 2018-03-07:
> I do not understand which part of my code is wrong....

This is beyond the scope of this issue tracker.
It looks like you are using the wrong version of some files.
If you rely on other packages, they might have changed upstream.

> I tried to set 'VERBOSE_OUTPUT=true` but, it does not works....

As travis suggests, you can increase the timeout:
`script:  travis_wait 30 .ci_config/travis.sh`

ipa-mdl, 2018-03-07:
> It looks like you are using the wrong version of some files.

Your packages misses some  `<run_depend>` lines.
If you need to find something, you depend on it.

RyodoTanaka, 2018-03-08:
@ipa-mdl 
### Try 1
> Your packages misses some <run_depend> lines.
> If you need to find something, you depend on it.

I checked again the dependency, and fixed what I missed at following.
 https://github.com/CIR-KIT/fourth_robot_pkg/commit/f6ec4dbe9788ff31223275b2cc834c542a263183 
But, the travis still returns the error.
https://travis-ci.org/CIR-KIT/fourth_robot_pkg/builds/350808396

### Try 2
To read [the error message](https://travis-ci.org/CIR-KIT/fourth_robot_pkg/builds/350808396#L2217-L2345), it seems that the compiler could not find my `fourth_robot_description` package....
And from the message, the compiler is finding in the catkin install directory.
That' why, I add installation part on my CMakeLists.txt of `fourth_robot_description` at following.
https://github.com/CIR-KIT/fourth_robot_pkg/commit/4b8b9a827fd8db5361acb48ca1071c62f2e6bbcd 
But, the travis still returns same error...
https://travis-ci.org/CIR-KIT/fourth_robot_pkg/builds/350826313

RyodoTanaka, 2018-05-23:
@ipa-bnm 
Sorry for so long reply.
But, I succeed to build & test every thing.
Here is what I did.
#### Add Installation part on CMakeList.txt
I added Installation part on each package's CMakeList.txt as following.
```cmake
foreach(dir launch urdf)
  install(DIRECTORY ${dir}/
    DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/${dir})
endforeach(dir)
```
I referenced [RethinkRobotics/sawyer_robot description package's CMakeList.txt](https://github.com/RethinkRobotics/sawyer_robot/blob/master/sawyer_description/CMakeLists.txt).

And also I added Test part like following
```cmake
if (CATKIN_ENABLE_TESTING)
  find_package(roslaunch REQUIRED)
  roslaunch_add_file_check(launch)
endif()
```
Adding above, my problems are solved.
The following is the result of Travis.
https://travis-ci.org/CIR-KIT/fourth_robot_pkg/builds/382480564

I'm sorry to bother you about this kind of basic mistakes...
And Thank you so much for your kind help.

closed 2018-05-27 by RyodoTanaka