ros-industrial/industrial_ci #275: Multiple packages found with the same name "catkin_tools_prebuild" "industrial_ci"

amc-nu, 2018-04-27:
First of all, thanks for sharing your amazing project!

I can run industrial_ci with no problem on travis-ci. However, when I try to run it locally, I get the following error:
```
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function setup_rosws returned with code '0' after 0 min 0 sec 
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

ERROR: Rosdep experienced an error: Multiple packages found with the same name "catkin_tools_prebuild":
- Autoware/ros/build/catkin_tools_prebuild
- Autoware/ros/install/share/catkin_tools_prebuild
Multiple packages found with the same name "industrial_ci":
- Autoware/ros/install/share/industrial_ci
- Autoware/ros/src/industrial_ci/industrial_ci
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.2

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 443, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 95, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "catkin_tools_prebuild":
- Autoware/ros/build/catkin_tools_prebuild
- Autoware/ros/install/share/catkin_tools_prebuild
Multiple packages found with the same name "industrial_ci":
- Autoware/ros/install/share/industrial_ci
- Autoware/ros/src/industrial_ci/industrial_ci

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function rosdep_install returned with code '1' after 0 min 0 sec 
Tests failed exit code '1'
```

I'm running from this [repo](https://github.com/CPFL/Autoware/tree/feature/industrial_ci) with this [configuration file](https://github.com/CPFL/Autoware/blob/feature/industrial_ci/.travis.yml)

I tried running this on two setups Ubuntu 14.04.5 and Ubuntu 16.04.4 with the same result.

Could you please have a look and provide some guidance? 
Thanks in advance.

amc-nu, 2018-04-27:
Sorry guys, found my mistake. I was using the same workspace for industrial_ci and my project.

closed 2018-04-27 by amc-nu

ipa-mdl, 2018-04-27:
> I was using the same workspace for industrial_ci and my project.

This should not be a problem, but you should run it for your source space only. Or even better just for a single repository (as with travis).

Please note that `run_ci` takes the target repo as first argument and would use the current directory if none was provided.
Perhaps we should enforce specifiying this argument.

amc-nu, 2018-04-27:
yeah, I cloned industrial_ci to the same src space, that was the problem.
Moving it out and sourcing the terminal fixed the problem.
Thanks for your comment!