ros-industrial/industrial_ci #276: Process stalls after build completion

amc-nu, 2018-04-29:
I've been using industrial_ci locally with some projects tested successfully. However, with this [one](https://github.com/CPFL/Autoware/tree/feature/industrial_ci) the process stalls.

I'm executing it like this:
`rosrun industrial_ci run_ci ROS_DISTRO=indigo ROS_REPO=ros`
`rosrun industrial_ci run_ci ROS_DISTRO=kinetic ROS_REPO=ros`
(I also used the `rosrun industrial_ci run_travis` with the same result).

I've ran it in the same way with other smaller projects, and what I can see is that with Autoware, **the build summary never shows up**, the process stalls and the next step (`catkin_run_tests`) is never reached. I've let it run for for up to 36 hours in different setups with no success. There is no error message. The last message shown is when the last package gets built and installed. 
```
...

Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: ""
-- Installing: /root/catkin_ws/install/_setup_util.py
-- Installing: /root/catkin_ws/install/env.sh
-- Installing: /root/catkin_ws/install/setup.bash
-- Installing: /root/catkin_ws/install/setup.sh
-- Installing: /root/catkin_ws/install/setup.zsh
-- Installing: /root/catkin_ws/install/.rosinstall
-- Up-to-date: /root/catkin_ws/install/lib/pkgconfig/ndt_localizer.pc
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/cmake/ndt_localizerConfig.cmake
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/cmake/ndt_localizerConfig-version.cmake
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/package.xml
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/ndt_matching_tku
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/ndt_mapping_tku
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/ndt_mapping_tku
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/ndt_matching
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/ndt_mapping
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/approximate_ndt_mapping
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/tf_mapping
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/lazy_ndt_mapping
-- Up-to-date: /root/catkin_ws/install/lib/ndt_localizer/queue_counter
-- Installing: /root/catkin_ws/install/share/ndt_localizer/launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/approximate_ndt_mapping.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/ndt_mapping_tku.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/ndt_matching_tku.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/tf_mapping.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/lazy_ndt_mapping.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/ndt_matching.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/ndt_mapping.launch
-- Up-to-date: /root/catkin_ws/install/share/ndt_localizer/launch/ndt_matching_tku_org.launch
cd /root/catkin_ws/build/ndt_localizer; catkin build --get-env ndt_localizer | catkin env -si  /usr/bin/make install; cd -
Finished   << ndt_localizer:install
Finished  <<< ndt_localizer                                   [ 1.6 seconds ]
```
For the same reason, when I run this in Travis-CI, the process gets killed when the time limit is reached. https://travis-ci.org/CPFL/Autoware/jobs/371840751

I'll be grateful if have any ideas or suggestions.

ipa-mdl, 2018-04-30:
I ran `run_ci ROS_DISTRO=kinetic ROS_REPO=ros` locally and got
```
/root/catkin_ws/src/Autoware/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_matching/ndt_matching.cpp:74:40:
fatal error: pcl_omp/registration/ndt.h: No such file or directory
```
(The build took 22 minutes on my laptop)

I cannot say why it got stalled for you..
Do you have enough free disk space? Have you checked syslog for errors?
You could also try to start `run_ci` direclty (via full path, not `rosrun`) in a clean (no source'd ROS env)  terminal.

ipa-mdl, 2018-04-30:
In addtion `test_points_preprocessor` needs a running ROS master, but does not get started in a rostest environment.

amc-nu, 2018-04-30:
@ipa-mdl Thank you for your prompt response. I can see this error now. Moreover, I realized execution is way faster than `rosrun`.

> Do you have enough free disk space?
Yes, no problem here.

> Have you checked syslog for errors?
Yes, but no error.

Thanks!

ipa-mdl, 2018-04-30:
The problem with the missing header file seems to be a missing dependency, which leads to the wrong build order.
(it works if built in devel space instead of install space)

amc-nu, 2018-05-02:
Hello @ipa-mdl 
Thank you for your help last time.

I fixed the errors on that package, and ran again the whole thing. However, I'm getting again stuck.
Let me give you a wider idea of what I'm doing. Maybe you can quickly spot the problem.
Thanks in advance.
This is how I installed `industrial_ci`
```
~
├── Autoware (workspace I want to test)
│    ├── ros
│    │     └──src
│    └──.travis.yml
├── catkin_ws (here is industrial_ci)
     ├── src
     │     └──industrial_ci (clone from this repo)
```

I ran `industrial_ci` using the following command while in `~/Autoware` ( I didn't source as previously suggested).

`~/catkin_ws/src/industrial_ci/industrial_ci/scripts/run_ci  ROS_DISTRO=indigo ROS_REPO=ros`
and
`~/catkin_ws/src/industrial_ci/industrial_ci/scripts/run_ci  ROS_DISTRO=kinetic ROS_REPO=ros`

The compilation finishes, the tests start and seems to finish.

```
cd /root/catkin_ws/build/ndt_localizer; catkin build --get-env ndt_localizer | catkin env -si  /usr/bin/make install; cd -
Finished   << ndt_localizer:install
Finished  <<< ndt_localizer                                   [ 2.1 seconds ]
```

No error, no syslog output, disk has plenty of space.

---

this might be unrelated but running on trusty, seems to be several times slower (especially the part `Extracting templates from packages`). Have you seen this before?

ipa-mdl, 2018-05-02:
> The compilation finishes, the tests start and seems to finish.

The tests are stuck (not finished!) because [test_points_preprocessor still needs a running ROS master](https://github.com/ros-industrial/industrial_ci/issues/276#issuecomment-385440877).
Please wrap it with [rostest](http://wiki.ros.org/rostest/Writing). 
You can run the tests from your workspace with `catkin[_make] run_tests`.

ipa-mdl, 2018-05-02:
> this might be unrelated but running on trusty, seems to be several times slower (especially the part Extracting templates from packages). Have you seen this before?

This an ouptut from apt/dpkg (used by ` rosdep install`).
The `kinetic` might need less (transitive) dependencies.
If you want to speed-up this step, you could prepare your custom ` DOCKER_IMAGE` or use an exisiting one.

amc-nu, 2018-05-03:
Thanks @ipa-mdl !

amc-nu, 2018-05-07:
@ipa-mdl 
I decided to remove most of the test and now I'm able to finish all the steps locally. Running on both rosrun or directly the ci.sh script.

However, when running on Travis, the process gets stalled as well. I'm not sure if this is because of the travis time limit or because it really gets stalled.

Local Output:
```
[build] Summary: All 114 packages succeeded!
[build]   Ignored:   2 packages were skipped or are blacklisted.
[build]   Warnings:  None.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 1 minute and 49.0 seconds total.
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function catkin_build_tests returned with code '0' after 1 min 49 sec

[build] Summary: All 114 packages succeeded!
[build]   Ignored:   2 packages were skipped or are blacklisted.
[build]   Warnings:  2 packages succeeded with warnings.
[build]   Abandoned: None.
[build]   Failed:    None.
[build] Runtime: 2 minutes and 5.1 seconds total.
Summary: 66 tests, 0 errors, 0 failures, 0 skipped
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function catkin_run_tests returned with code '0' after 2 min 6 sec 
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function catkin_install_run_tests returned with code '0' after 0 min 0 sec 
All tests succeeded
```

Travis-CI Output:
```
CUDA_TOOLKIT_ROOT_DIR not found or specified
cd /root/catkin_ws/build/ndt_localizer; catkin build --get-env ndt_localizer | catkin env -si  /usr/bin/cmake /root/catkin_ws/src/Autoware/ros/src/computing/perception/localization/packages/ndt_localizer --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel/.private/ndt_localizer -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install; cd -
...............................................................................
No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
The build has been terminated
```

Could you please give me your opinion? ideas?

--
Different question, last time you kindly pointed the problem about the rostest not being wrapped in rostest. Can you please tell me how did you find out? is there any way to check where is the process stalled inside `industrial_ci`?

Thanks in advance.

ipa-mdl, 2018-05-09:
> 

You can increase the timeout with [travis_wait](https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received): `travis_wait .industrial_ci/travis.sh`.

> Different question, last time you kindly pointed the problem about the rostest not being wrapped in rostest. Can you please tell me how did you find out?

It was printed in the logs.

> is there any way to check where is the process stalled inside industrial_ci?

You can enter the running instance with `docker exec` or list the processes with `docker top`.

closed 2018-05-09 by amc-nu