ros-industrial/industrial_ci #281: Add travis_wait-like feature

ipa-mdl, 2018-05-29:
`travis_wait`  adds some  log message to the output at the end of each job, but might be required for installing a lot of packages or non-verbose build.
Instead we  could include a custom output that will be displayed each 9 minutes, like in the travis script I prepared for [ros-industrial/docker](https://github.com/ros-industrial/docker/blob/master/build.sh#L17).

This would make travis run as long as needed, but there is a little chance that the output (especially the folds) gets mangled.

fmessmer, 2018-05-30:
I happen to see the timeout message quite a lot recently:
```
No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
The build has been terminated
```

Also, for jobs that have been working before.
It seems to happen for both `indigo` and `kinetic` jobs.
Has anything changed in `industrial_ci` or `travis` itself recently that causes the increase in build terminations?

ipa-mdl, 2018-06-08:
The `industrial_ci ` sporadically show the same issue now, in jobs that should not exceep 3 minutes..

ipa-mdl, 2018-11-16:
implemented in #324

closed 2018-11-16 by ipa-mdl