ros-industrial/industrial_ci #295: gitlab runner hangs on wstool merge

ipa-fez, 2018-07-05:
Job output:

```
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function setup_rosdep returned with code '0' after 0 min 16 sec 
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Writing /root/catkin_ws/src/.rosinstall

update complete.

     Change details of element (Use --merge-keep or --merge-replace to change):
  repo_1

     Add new elements:
  repo_2,  repo_3,  repo_4

Continue: (y)es, (n)o, (v)erbosity, (a)dvanced options: 
```

It waits for input at this point until the job times out.

Our .gitlab-ci.yaml:

```
image: docker:git # docker and git clients

variables:
  TMPDIR: "${CI_PROJECT_DIR}.tmp" #
  DOCKER_DRIVER: "overlay2" # recommended for docker-in-docker, needs overlay kernel module loaded on host

# enable docker-in-docker
services:
  - docker:dind

before_script:
  - apk add --update bash coreutils tar # install industrial_ci dependencies
  - git clone https://github.com/ros-industrial/industrial_ci .ci_config

# setup the actual tests

kinetic:
  script: .ci_config/gitlab.sh
  variables:
    ROS_DISTRO: "kinetic"
    UPSTREAM_WORKSPACE: "file"
```

This is currently happening on a merge-request. Restarting the job does not resolve the issue. The master branch was building fine, but there it has all repos in the "Add new elements" line.

Another difference seems to be that the merge request output starts with "Cloning repository..." while the master branch says "Fetching changes" instead.

A workaround could be adding a `-y` flag to the `wstool merge`.

ipa-mdl, 2018-07-05:
Thanks for reporting!
Obviously, something seems to be wrong with the rosinstall files or how they get processed.
I would expect `/root/catkin_ws/src/.rosinstall` to be empty before the merge (based on you config).

Can you provide an example that can be tested with `run_ci`?
Or grant me access to your Gitlab instance..

> Another difference seems to be that the merge request output starts with "Cloning repository..." while the master branch says "Fetching changes" instead.

This is completely unrelated.

> A workaround could be adding a -y flag to the wstool merge.

I am not sure about the other side effects.

ipa-mdl, 2018-07-06:
I was able to replicate this error with my runner.
Next step is to test an example on gitlab.com.

ipa-mdl, 2018-07-06:
This might have been an issue with the content of the rosinstall file, it contained a duplicated `local-name`.
@ipa-fez: Please fix and test again.

ipa-fez, 2018-07-09:
Yes, that was it. I've fixed it and now it's running as expected. Thanks for your help!

I'll close this issue.

closed 2018-07-09 by ipa-fez