ros-industrial/industrial_ci #296: Defined but empty  'DOCKER_IMAGE' does not cause error.

130s, 2018-07-09:
When passing empty `DOCKER_IMAGE` variable, CI continues although the intended docker image wasn't pulled. This makes debugging harder.

ipa-mdl, 2018-07-09:
Can you give more context, please?
`DOCKER_IMAGE`  has multiple purposes, it might be used differently with regard to other settings.
In addition it would be good to know what the error looks like.

ipa-mdl, 2018-07-10:
I see.
This case should be fixed [here](https://github.com/ros-industrial/industrial_ci/blob/d66bf0f1578008c23f0673ff8894e20308f067c0/industrial_ci/src/docker.sh#L167), because it just tests the length..

machinekoder, 2018-07-11:
Working on this for #wrid18

130s, 2018-07-11:
Maybe related to #212 btw.

ipa-mdl, 2018-07-11:
#212 is not related.

130s, 2018-07-11:
> #212 is not related.

You're right (I'm blaming my bad view on github from cellphone :/ ).

ipa-mdl, 2019-12-04:
Reviewing this issue, I would now opt for "wont fix".
`DOCKER_IMAGE=''` was not handled as error, because it means "use the default".
If we check this, then we have to a lot more variables.

PR #617 (2021-02-07): Improve DOCKER_IMAGE detection
closes #296
closes #299
closed 2021-02-07 by ipa-mdl