ros-industrial/industrial_ci #298: Tests in "test_depend"-ed pkgs shouldn't run?

130s, 2018-07-11:
I haven't found a way to specify the packages that are `test_depend`-ed and need to be built, but don't want to run the tests of.

What I tried (with a risk that this may be wrong) is to clone test_depended pkgs via .rosinstall, add them in `BUILD_PKGS_WHITELIST` so that they get built.
With this, I see tests in `test_depend`ed packages are run in `catkin_run_tests`, which I do not want.

This might be the scope in https://github.com/ros-industrial/industrial_ci/pull/137, but that looks to be aiming larger than just this, beside it is unmerged yet.

ipa-mdl, 2018-07-11:
Can you give an example?
`test_depend`-ed packages will to be built if needed, but, tests won't be run.

130s, 2018-07-11:
I don't have a public example for now, so only verbally tracks the logic.

1. Pkgs in `BUILD_PKGS_WHITELIST` get added into `PKGS_DOWNSTREAM` https://github.com/ros-industrial/industrial_ci/blob/d375a781f6f7ffdfb71e0abb4e4a47e0be309bd1/industrial_ci/src/tests/source_tests.sh#L177
2. Pkgs in `PKGS_DOWNSTREAM` are:
  -  `catkin run_tests`-ed https://github.com/ros-industrial/industrial_ci/blob/d375a781f6f7ffdfb71e0abb4e4a47e0be309bd1/industrial_ci/src/tests/source_tests.sh#L197
  - `rostest`-ed https://github.com/ros-industrial/industrial_ci/blob/d375a781f6f7ffdfb71e0abb4e4a47e0be309bd1/industrial_ci/src/tests/source_tests.sh#L220

I might be misunderstanding something, but due to the lines above, currently the tests in pkgs in `BUILD_PKGS_WHITELIST` are supposed to be run (which I wonder if there's an option to disable).

130s, 2018-07-11:
Closing as I think the usecase is the same as discussed in https://github.com/ros-industrial/industrial_ci/issues/172 (and by starting to ask about test_depend I might have been creating XY problem).

closed 2018-07-11 by 130s