ros-industrial/industrial_ci #306: Cannot skip rosdep step

machinekoder, 2018-07-13:
In some cases, e.g. when using an already prepared Docker image, we can skip the `rosdep install` step. This is especially useful when our dependencies aren't released yet.

machinekoder, 2018-07-13:
Alternatively, overwriting the `rosdep` command would also help. Moreover, it may be useful to execute the `wstool init` step before the `rosdep` step, since this might add additional dependencies.

machinekoder, 2018-07-13:
Ignore my last statement, `wstool` is executed before the `rosdep install` step.

machinekoder, 2018-07-13:
I think `rosdep` is missing the `--os=debian:stretch` option.

ipa-mdl, 2018-07-13:
> In some cases, e.g. when using an already prepared Docker image, we can skip the rosdep install step. This is especially useful when our dependencies aren't released yet.

Why do you want to skip rosdep in this case?
In addition to installing them, `rosdep` checks that the dependencies exist and can be resolved.
However, I could imagine to skip `rosdep update`.

> I think rosdep is missing the --os=debian:stretch option.

rosdep auto-detects the OS. You could override the detection by setting `ROS_OS_OVERRIDE`.

machinekoder, 2018-07-13:
Then the OS autodetection does not work correctly. It looks for `stretch` rather than `debian:stretch`. The rosdep works perfectly fine when passing `--os` manually.

ipa-mdl, 2018-07-13:
If rosdep fails to detect Debian stretch, this should better be fix in `rosdep`/`rospkg`.
Which error do you get?

machinekoder, 2018-07-13:
I'm doing something wrong, I did not specify `UPSTREAM_WORKSPACE` to select a file for the `.rosinstall`.

closed 2018-07-13 by machinekoder