ros-industrial/industrial_ci #308: rostests with arguments failing in install space

mgruhler, 2018-07-16:
I've found that `rostest`s in the install space fail when you try to pass arguments to them.
This is reproducible both locally and on travis.

I created a minimal-example repo [here](https://github.com/mgruhler/industrial_ci_rostest) (with the test file stolen from the `rostest` package).
As you can see, [build 1](https://travis-ci.org/mgruhler/industrial_ci_rostest/builds/404370214) with `NO_TEST_INSTALL=true` works fine, but [build 2](https://travis-ci.org/mgruhler/industrial_ci_rostest/builds/404388903) with [this line removed](https://github.com/mgruhler/industrial_ci_rostest/commit/aec861b6140cd911ccb8187cd22a84692d0d0603) fails and complains about that it `requires the 'topic' arg to be set`.

Apparently, argument passing doesn't work for `rostest`s in the install space.

I've found the following related issue #150 which led me to the cob_* repositories, where the apparent solution is to skip all tests for install spaces (see e.g. ipa320/cob_robots#638).

Is there any reason why this cannot work that I missed? Or is this something that should be implemented in the future?

Thx.

ipa-mdl, 2018-07-16:
TL;DR: IMHO `NO_TEST_INSTALL=true` should be the default (and the option renamed..).

I am not sure why this install rostest feature was added in the first place, but it is now in for backwards compatibilty only.
Your test gets run in install space as part of the `catkin_run_tests` step. This step includes the arguments and succeeds.

The failing `catkin_install_run_tests` does something else:
It crawls the install space for *.test files and runs them with `rostest`.
There is no way to specifiy any arguments, because these need to be specified for each invocation individually. (=> travis yml nightmare)

In your case it should be fine to set `NO_TEST_INSTALL=true` (like I did in https://github.com/ipa320/cob_robots/pull/638).
I cannot imagine any useful scenario for `NO_TEST_INSTALL=false` (given that install space is the default target).
In most cases it would be much better to test each *.test file in `run_tests` (as you did).

mgruhler, 2018-07-16:
> Your test gets run in install space as part of the catkin_run_tests step. This step includes the arguments and succeeds.

Do you mean in the `devel` space? Otherwise, I'm still not sure I understand the difference between the two `catkin*run_tests` invocations. The one runs all tests and the other just searches `*.test` files and tests those?

ipa-mdl, 2018-07-16:
> Do you mean in the devel space?

No, the devel space is not used (per default).

> The one runs all tests 

Yes, but dependencies have to be present in install space.

> The one runs all tests and the other just searches *.test files and tests those?

In this case "test" means ["invoke rostest"](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/tests/source_tests.sh#L228)

mgruhler, 2018-07-17:
Now I understand. Thanks for explaining @ipa-mdl 

The latter is actually "manually" calling `rostest` on any installed `*.test` file.

I agree that this invocation is something unexpacted and thus this should not be the default. I.e. `NOT_TEST_ISNTALL=true` would be the better way to go (with renaming the very confusing variable name).

But as this is more a "misunderstanding" from my side, I'm fine with the explanation and close this issue.
If you feel that this would be worth taking up action, please reopen.

closed 2018-07-17 by mgruhler