ros-industrial/industrial_ci #312: run_ci fails with "Error: image X not found"

130s, 2018-08-09:
I have a private repo on Gitlab where the Docker image is registered. I assume this fails because the "base" docker image run by industrial_ci does not yet run `docker login`. If so, providing a login feature is nice.
```
$ rosrun industrial_ci run_ci DOCKER_IMAGE=repooo/userr:tagg ROS_DISTRO="indigo"
Testing /home/n130s/cws_proddd_k/src/project-y/project-y
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Pulling repository registry.gitlab.com/userr/project-y
Error: image userr/project-y:tagg not found
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function prepare_docker_image returned with code '1' after 0 min 10 sec 
Tests failed exit code '1'
```

ipa-mdl, 2018-12-17:
I don't see the need for a login function in `industrial_ci`.
Just login before starting `run_ci`, Docker will save the password for future uses..

ipa-mdl, 2020-04-08:
I will close this for now.
`run_ci`  is meant as a thin wrapper and should not deal with credentials etc.

closed 2020-04-08 by ipa-mdl