ros-industrial/industrial_ci #314: docker-in-docker concern for Gitlab CI

130s, 2018-08-10:
Moved from https://discourse.ros.org/t/industrial-ci-for-gitlab-using-docker-in-docker-reason/5684

> Hi! I have a hard time understanding why the industrial_ci for gitlab (https://github.com/ros-industrial/industrial_ci#for-gitlab-ci 3)
> 
> uses the Docker-in-Docker technique which gave me quite a hard time behind the cooperate proxy and I still don’t have it up and working.
> 
> Wouldn’t a shell executor be sufficient? Or could there be issues with proper cleanup?

130s, 2018-08-10:
Thank you for the concern. 

What is your problem exactly? I assume you're seeing an issue that you think is rooted in proxy, but without seeing description and/or the system output, it's hard to provide any useful feedback.

You mentioned concern for docker-in-docker though at the moment, I don't know if that's what's causing issues for you. That said,
my opinion in this regard is that adding extra layers (e.g. docker-in-docker) frees the tool from caring the platform specifics. For instance, using container we don't need to care `Gitlab CI`'s .yml shell limitation. @ipa-mdl is the best person and he should have more thorough rationale.

ipa-mdl, 2018-08-10:
@agutenkunst: 
You don't have to use docker-in-docker. This is just the only supported way of running Docker on gitlab.com shared runners.
As an alternative you could use the Docker executor with a shared `/var/run/docker.sock`.
A shell executor (which can run Docker) mght work as well, but I haven't tested it yet because of security concerns..
(https://docs.gitlab.com/ee/ci/docker/using_docker_build.html)

Regarding your proxy problem: Have you tried [this](https://docs.docker.com/network/proxy/#configure-the-docker-client)?

agutenkunst, 2018-08-13:
I was able to boil the problem down to:

Dockerfile
```
FROM ubuntu:latest

RUN echo "test"
```
.gitlab-ci.yml
```
image: docker:git

variables:
  DOCKER_DRIVER: overlay2

services:
  - docker:dind
indigo:
  script:
    - echo $HTTP_PROXY
    - echo $HTTPS_PROXY
    - echo $http_proxy
    - echo $https_proxy
    - docker build --build-arg https_proxy=http://172.16.206.33:8080 .
```

But I get:
```
Running with gitlab-runner 11.1.0 (081978aa)
  on alex_vm_docker 6c9f3e72
Using Docker executor with image docker:git ...
Starting service docker:dind ...
Pulling docker image docker:dind ...
Using docker image sha256:1e09f740386c351baa55960c86f34c7f64ca7ded041ea832eabe518d5c63c14b for docker:dind ...
Waiting for services to be up and running...
Pulling docker image docker:git ...
Using docker image sha256:b66a9cf1756f130b83e900b460112b5835fea127df51ed29128b04f42261bfcb for docker:git ...
Running on runner-6c9f3e72-project-157-concurrent-0 via runner...
Fetching changes...
Removing .industrial_ci/
HEAD is now at 17f00f2 Debug
From http://srv-git01.de.pilz.local:9999/ROS-PLWA/pilz_robots
 + 492852d...5689e8a feature/gitlab -> origin/feature/gitlab  (forced update)
Checking out 5689e8a5 as feature/gitlab...
Skipping Git submodules setup
$ echo $HTTP_PROXY

$ echo $HTTPS_PROXY

$ echo $http_proxy

$ echo $https_proxy

$ docker build --build-arg https_proxy=http://172.16.206.33:8080 .
Sending build context to Docker daemon  7.866MB

Step 1/2 : FROM ubuntu:latest
Get https://registry-1.docker.io/v2/: net/http: request canceled while waiting for connection (Client.Timeout exceeded while awaiting headers)
ERROR: Job failed: exit code 1
```

Could also be a DNS Issue...

ipa-mdl, 2021-07-21:
docker-in-docker is not needed anymore (for most jobs) and there is no need for image builds anymore.

closed 2021-07-21 by ipa-mdl