ros-industrial/industrial_ci #322: ABI check feature failing without obvious output but "ERROR: no symbols info in the ABI dump"

130s, 2018-10-20:
Of [this PR](https://github.com/ros-drivers/openni2_camera/pull/89), ABI check failed e.g. https://travis-ci.org/ros-drivers/openni2_camera/jobs/437001338
```
Preparing, please wait ...
ERROR: no symbols info in the ABI dump
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function abi_check_libopenni2_camera_nodelet returned with code '7' after 0 min 0 sec 
```
But other than that I failed to find the root cause of the error. There's a lot of warnings during the previous ABI check steps, but I don't see anything obviously failing.

130s, 2018-10-20:
Looks like error code 7 is `Invalid input ABI dump` according to the [abi-compliance-checker's doc](https://lvc.github.io/abi-compliance-checker/#Error).

ipa-mdl, 2018-10-20:
The big question is: Why is the ABI dump empty?
The build shows a lot of warnings.

I would suggest to test it locally and bind-mount the directories. This way you can introspect the results.
Or we could output the dump if the exit code is > 1.

ipa-mdl, 2018-10-20:
How is this different from https://github.com/ros-industrial/industrial_ci/issues/258?

PR #369 (2019-05-22): Backport some colcon features
Partial backport of #361 

* align the code/behavior between both version
* shellcheck :)
*  fix #258, fix #322 
* backport of CMAKE_ARGS (highly perferred over `CATKIN_CONFIG`)

In addition it should simplify the review.

# CoS (Condition of Satisfaction, suggested by @130s)
- Code review.
  - [x] @miguelprada 
  - [x] @130s
- (Locally) verify the change.
  - [x] @130s ABI error https://github.com/ros-industrial/industrial_ci/issues/258 -> Ongoing in https://github.com/130s/openni2_camera/pull/2
closed 2019-05-22 by 130s