ros-industrial/industrial_ci #323: ABI check feature's error output is not part of CI output

130s, 2018-10-20:
ABI check returns error in https://github.com/ros-drivers/openni2_camera/pull/78, without providing obvious clue. It's a bit difficult to debug from the current output.

E.g. https://travis-ci.org/ros-drivers/openni2_camera/jobs/359405485
```
Function abi_check_libopenni2_wrapper returned with code '0' after 0 min 0 sec 
Broken libraries: libopenni2_driver_lib
```

130s, 2018-10-20:
As @roehling said at https://github.com/ros-drivers/openni2_camera/pull/78#issuecomment-395689604, maybe include the output html file in CI output when ABI check ends with error?

ipa-mdl, 2018-10-20:
I don't understand this issue. The ABI report is given in the section above and it cleary states a binary compatbility of only 92.6%.

130s, 2018-10-20:
I think what https://github.com/ros-drivers/openni2_camera/pull/78#issuecomment-395689604 says is that the information about what part and files causes the incompatibility will help to improve the code to avoid ABI incompatibility (or make other decisions e.g. not to maintain it compatible). I'm not familiar with this ABI checker yet but does the current output provide that info in the CI output?

ipa-mdl, 2018-10-20:
The HMTL (summary) report gets dumped to the console.
If I recall it correctly, there are more files that could be dumped, but this could easily exceed the log line limit.
I would suggest to run it locally and mount the report folder for now.
(Something that might be mentioned in our documentation)
This why you might find out which files we could print in addition.