ros-industrial/industrial_ci #333: ROS2 support

ipa-jba, 2018-11-16:
Do you have any plans to make industrial_ci available for ROS2 (maybe related to #301)

ipa-mdl, 2018-11-16:
No actual plan yet.
I would be in favor of a colcon-based rewrite, perhaps in Python.

ipa-mdl, 2018-11-16:
I have now created a project: https://github.com/ros-industrial/industrial_ci/projects/1
For simplicity, the current version should be refactored first.

AustinDeric, 2018-12-03:
I use this script for ROS2 CI: https://github.com/AustinDeric/control_msgs/blob/bouncy-devel/.travis.yml
```
ervices:
  - docker

script:
- docker run --name build-container -dit osrf/ros2:bouncy-ros-base /bin/bash
- docker exec build-container /bin/bash -c 'mkdir -p /ros2_ws/src'
- docker exec build-container /bin/bash -c 'apt update && apt install -y python3-colcon-common-extensions python-rosdep && rosdep init && rosdep update'
- docker cp $TRAVIS_BUILD_DIR build-container:/ros2_ws/src/
- docker exec build-container /bin/bash -c 'source /opt/ros/bouncy/setup.bash && cd /ros2_ws/ && rosdep install --from-paths src --ignore-src --rosdistro bouncy -r -y'
- docker exec build-container /bin/bash -c 'source /opt/ros/bouncy/setup.bash && cd /ros2_ws/ && colcon build --symlink-install'
```
It works well:
https://travis-ci.org/AustinDeric/control_msgs/jobs/462963429

There is also a gitlab version that produces a  dockerfile:
https://github.com/AustinDeric/control_msgs/blob/bouncy-devel/.gitlab-ci.yml
```
image: docker:git
services:
  - docker:dind
variables:
  ENV_IMAGE: osrf/ros2:bouncy-ros-base
  BUILD_IMAGE: $CI_REGISTRY_IMAGE:$CI_COMMIT_REF_SLUG
stages:
  - build

build:
  stage: build
  before_script:
    - docker login -u gitlab-ci-token -p $CI_JOB_TOKEN $CI_REGISTRY
  script:
    - docker run --name build-container -dit $ENV_IMAGE /bin/bash
    - docker exec build-container /bin/bash -c 'mkdir -p /ros2_ws/src'
    - docker exec build-container /bin/bash -c 'apt update && apt install -y python3-colcon-common-extensions python-rosdep && rosdep init && rosdep update'
    - docker cp $CI_PROJECT_DIR build-container:/ros2_ws/src/
    - docker exec -w /ros2_ws build-container /bin/bash -c 'source /opt/ros/bouncy/setup.bash && rosdep install --from-paths src --ignore-src --rosdistro bouncy -r -y'
    - docker exec -w /ros2_ws build-container /bin/bash -c 'source /opt/ros/bouncy/setup.bash && colcon build --symlink-install'
    - docker commit build-container $BUILD_IMAGE
    - docker push $BUILD_IMAGE
```

I think industrial CI might want to leverage this approach.  Food for thought. I can submit an PR if interested?

ipa-mdl, 2018-12-03:
> I think industrial CI might want to leverage this approach.

I don't want to maintain two versions..
That's why I proposed to consolidate the current version first and switch to colcon.

The steps in your config do not run the tests, but it's more or less what we are doing right now (with catkin-tools).

130s, 2019-01-07:
Some analysis made by a user https://github.com/CPFL/Autoware/pull/1832#issuecomment-451900057

esteve, 2019-02-20:
Would it make sense to first add support for `colcon` and then plan the rewrite just to get things moving? We're using `industrial_ci` at https://github.com/CPFL/Autoware, however we'll soon add packages for ROS2 and thus remove `catkin` and use `colcon` for our builds instead, so unfortunately we'll have to resort to a bunch of custom scripts we wrote.

gbiggs, 2019-02-20:
How much effort is a rewrite in Python expected to require? What are the benefits beyond just adding `colcon` support first?

ipa-mdl, 2019-02-21:
> Would it make sense to first add support for `colcon` and then plan the rewrite just to get things moving?

Yes, this is/was my plan..
I have now a proof-of-concept (for colcon and ROS2): https://github.com/ipa-mdl/industrial_ci/tree/colcon
The workflow is a little bit different (proper upstream and downstream handling) and I dropped some features for now.

> How much effort is a rewrite in Python expected to require? What are the benefits beyond just adding `colcon` support first?

I am not sure how much effort it would be. The python migration would be benefical in using the python APIs of the build tools instead of the limited CLI.

ipa-mdl, 2019-02-22:
I addressed most items in https://github.com/ros-industrial/industrial_ci/projects/1

It seems to work for ROS1 (catkin_tools) and ROS2 (colcon): https://travis-ci.org/ipa-mdl/industrial_ci/builds/497145936
(The error are actual errors in the target code)

esteve, 2019-02-26:
@ipa-mdl thanks for the quick response. I'm going to give your branch (https://github.com/ipa-mdl/industrial_ci/tree/colcon) a try on Autoware and will report back. We're already using `colcon` with ROS1, since we'll start soon having a mixed workspace with packages written for ROS1 and ROS2.

ipa-mdl, 2019-02-26:
@esteve: For ROS1, the new version uses `catkin_tools`  per default, but you can set `BUILDER=colcon` (melodic only, because of the required `setuptools` version).
It took me some time to figure that it is bad idea to install `python3-rosdep` ;)

> since we'll start soon having a mixed workspace with packages written for ROS1 and ROS2.

I have already prepared basic support, but I don't know how to mix both properly (w.r.t. name collisions of message packages etc.).

amc-nu, 2019-03-29:
@ipa-mdl I confirmed your `colcon` branch works flawlessly with current Autoware `feature/melodic` branch.
https://travis-ci.org/amc-nu/Autoware/builds/512861001.
Thanks so much!

ipa-mdl, 2019-03-29:
@amc-nu: Thanks for testing!

Some comments on your config:
I have removed the explicit support for `CATKIN_PARALLEL_JOBS='-p2' ROS_PARALLEL_JOBS='-j2'`, so these values don't get passed as arguments anymore. Did you notice any difference?
Why do you set `_GUARD_INTERVAL=10`, this results in a message every 10 **seconds**.
This variable is mostly meant for testing, the default (540, 9 minutes) was too long for me to wait for ;)

amc-nu, 2019-03-29:
> I have removed the explicit support for CATKIN_PARALLEL_JOBS='-p2'

I saw colcon was still building running only two processes simulteanously. Not setting this, would cause a memory exhaustion problem on Travis while compiling.

> Why do you set _GUARD_INTERVAL=10

I required this for `catkin_tools`. Travis CI was killing our builds since no output was received for periods longer than 10 minutes. This is probably not required for `colcon`. It often prints `[processing ...]`

ipa-mdl, 2019-04-01:
>>  Why do you set _GUARD_INTERVAL=10

>I required this for catkin_tools. Travis CI was killing our builds since no output was received for periods longer than 10 minutes. 

"...industrial_ci is still running..." should get printed every 9 minutes to avoid the timeout.
If this does not work, please file an issue.

miguelprada, 2019-04-08:
I've ran some tests with this branch, both in our private repositories using GitLab-CI and on some Github repositories using Travis-CI.

In our private repos, jobs are failing with:

```
/root/ici/workspace.sh: line 30: exec: eval: not found
```

This error seems to be triggered by [this line](https://github.com/ipa-mdl/industrial_ci/blob/f01fc46b06c3ff06f7569267e989cc6f921a8ff3/industrial_ci/src/builders/catkin_tools.sh#L36) (we are passing some CMake arguments using `CATKIN_CONFIG`). I'm not sure why this happens...

The other tests I ran were on the UR-related repositories. Regular tests are succeeding but the prerelease tests fail. See:

- [universal_robot](https://travis-ci.org/miguelprada/universal_robot/builds/516709185)
- [ur_modern_driver](https://travis-ci.org/miguelprada/ur_modern_driver/builds/516709942) (disregard failing jobs, those depend on unreleased features in `universal_robot`)

I checked whether the prelease test was failing just for the absence of the apt key store-related variables, but [more problems show up down the line](https://travis-ci.org/miguelprada/universal_robot/jobs/516719342).

> @esteve: For ROS1, the new version uses catkin_tools per default, but you can set BUILDER=colcon (melodic only, because of the required setuptools version).

Not very familiar with `colcon` yet. Does this mean that `colcon` doesn't support Xenial at all?

miguelprada, 2019-04-08:
Also, our internal jobs use `BEFORE_SCRIPT` to perform some custom setup steps, and I believe this is not handled in this branch. Are there any plans to restore this functionality?

ipa-mdl, 2019-04-08:
@miguelprada: Thanks for testing!

The handling of `CATKIN_CONFIG` seems to be a bug.
I did not touch the pre-release tests (?), I will check it.

> Also, our internal jobs use BEFORE_SCRIPT to perform some custom setup steps, and I believe this is not handled in this branch. Are there any plans to restore this functionality?

I can bring it back, because the exact step does not exists anymore.
As an alternative, you can now add scripts before/after all steps.
For example to run after `setup_target_workspace`, just set `AFTER_SETUP_TARGET_WORKSPACE`.

> Does this mean that colcon doesn't support Xenial at all?

It works, but it is not well supported.

miguelprada, 2019-04-08:
> I did not touch the pre-release tests (?), I will check it.

I think the error may be related to [`ROS_REPOSITORY_PATH` not being set](https://github.com/ipa-mdl/industrial_ci/blob/f01fc46b06c3ff06f7569267e989cc6f921a8ff3/industrial_ci/src/tests/ros_prerelease.sh#L43) anymore by default.

> As an alternative, you can now add scripts before/after all steps.
For example to run after setup_target_workspace, just set AFTER_SETUP_TARGET_WORKSPACE.

Understood. Would it make sense to add a deprecation warning and setting `AFTER_SETUP_TARGET_WORKSPACE` with `BEFORE_SCRIPT` for backward compatibility?

ipa-mdl, 2019-04-08:
> Would it make sense to add a deprecation warning

Definately!
Perhaps it should even raise an error.

> setting AFTER_SETUP_TARGET_WORKSPACE with BEFORE_SCRIPT for backward compatibility?

No, this is not really  backwards compatible. In addtition paths have changed, so the scripts will have get changes most likely.
Instead we should provide a migration guide.

closed 2019-08-18 by ipa-mdl