ros-industrial/industrial_ci #336: [Melodic/Bionic] Docker, command not found

JimmyDaSilva, 2018-11-28:
Hello, 

I am trying to compile as well on Melodic and Ubuntu 18.04, but I get a `docker not found` in the `docker.sh`

Here is the job log:
```
$ source .industrial_ci/travis.sh
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Building image 'industrial-ci/melodic/ubuntu:bionic':
FROM ubuntu:bionic
RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
RUN apt-get update -qq     && apt-get -qq install --no-install-recommends -y apt-utils gnupg wget ca-certificates lsb-release
RUN echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list
RUN apt-key adv --keyserver "hkp://ha.pool.sks-keyservers.net" --recv-key "0xB01FA116"     || { wget "https://raw.githubusercontent.com/ros/rosdistro/master/ros.key" -O - | apt-key add -; }
RUN sed -i "/^# deb.*multiverse/ s/^# //" /etc/apt/sources.list     && apt-get update -qq     && apt-get -qq install --no-install-recommends -y         build-essential         python-catkin-tools         python-pip         python-rosdep         python-wstool         ros-melodic-catkin         ssh-client     && apt-get clean     && rm -rf /var/lib/apt/lists/*
/home/travis/build/JimmyDaSilva/rtt_lwr/.industrial_ci/industrial_ci/src/docker.sh: line 146: docker: command not found
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function prepare_docker_image returned with code '127' after 0 min 0 sec 
The command "source .industrial_ci/travis.sh" exited with 127.
```

Many thanks ! 
Jimmy

ipa-mdl, 2018-11-28:
You config seems to be wrong.
Please have a look at some sample config like [ros_controllers](https://github.com/ros-controls/ros_controllers/blob/melodic-devel/.travis.yml).

JimmyDaSilva, 2018-11-28:
Thanks @ipa-mdl .
For other users, the problem was that it need to be specified as a trusty dist, even if it's for Ubuntu 18.04 which is called bionic. Very confusing ...

`dist: trusty` solves the issue

closed 2018-11-28 by JimmyDaSilva

ipa-mdl, 2018-11-28:
> even if it's for Ubuntu 18.04 which is called bionic. Very confusing ...

Travis does not provide a 18.04 environment. That's why we are using Docker under the hood.

JimmyDaSilva, 2018-11-29:
I see, it makes perfect sense.