ros-industrial/industrial_ci #339: Specify clang-format-check version

wxmerkt, 2018-12-08:
The clang-format-check added in #320 uses the system-default clang-format (which changes between operating system versions). It would be great if the version could be specified.

ipa-mdl, 2018-12-09:
You can specify the version indirectly by changing `ROS_DISTRO. For now it is required to bet set it, but is only used for selecting a Docker image.

Which version do you want to use? And how to install it?

wxmerkt, 2018-12-09:
We noticed differences in formatting between versions and thus want to specify what the agreed upon standard version is - across distros (we support indigo, kinetic, melodic). Eg for kinetic and melodic we want to specify CLANG_FORMAT_EXECUTABLE=clang-format-3.9 - with this version apt-get installed and used in the formatting command.

ipa-mdl, 2018-12-09:
The problem is that not all `clang-format` versions are availailable for all Ubuntu versions..
However, I just found out that there is no `clang-format` alias for `indigo`/`trusty`.

In general `CLANG_FORMAT_EXECUTABLE` makes sense to me, espcially if we default it to `clang-format`.
It would be great if you could prepare a pull request for this.

ipa-mdl, 2018-12-10:
fixed in #340

closed 2018-12-10 by ipa-mdl