ros-industrial/industrial_ci #342: Clang Format blacklist directories?

Levi-Armstrong, 2018-12-17:
In one of my repositories I have sub-modules that get pulled in from third party repositories that I would like to blacklist from the clang format check. Is it possible to blacklist or whitelist packages for the clang-format test. If not would the maintainers be supportive of adding this feature, I would be happy add this feature?

ipa-mdl, 2018-12-17:
At the moment there is no blacklist/whitelist option for the clang format check.
Perhaps you can [disable submodules cloning](https://docs.travis-ci.com/user/customizing-the-build/#git-submodules) for a specific job.

> If not would the maintainers be supportive of adding this feature, I would be happy add this feature?

Of course!
The file selection happens around [here](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/tests/clang_format_check.sh#L44).

I could imaging to implement some wildcard matching against "file", but I haven't found a nice solution yet.

Levi-Armstrong, 2018-12-18:
I have something working providing package/directory name. The example below would exclude file that contain any directory in the path that is listed int TEST_BLACKLIST.

TEST_BLACKLIST=stomp,industrial_core

``` bash
   CLANG_PKGS=$((echo "$TEST_BLACKLIST" | tr ' ;,' '\n') | tr '\n' ' ')
   if [ -n "$CLANG_PKGS" ]; then
       IFS=' ' read -r -a CLANG_PKGS <<< "$CLANG_PKGS"
   fi
 
   PRUNE=""
   CNT=0
   for i in "${CLANG_PKGS[@]}"
   do
       if [[ $CNT -eq 0 ]]; then
           PRUNE="-name $i"
       else
           PRUNE="$PRUNE -o -name $i"
       fi
       let CNT+=1
   done
 
   # Run clang-format
   if [[ $CNT -eq 0 ]]; then
       find . -type f -regex '.*\.\(cpp\|hpp\|cc\|cxx\|h\|hxx\)' -exec clang-format -style=file -i {} \;
   else
       find . -type d \( $PRUNE \) -prune -o -type f -regex '.*\.\(cpp\|hpp\|cc\|cxx\|h\|hxx\)' -exec clang-format -style=file -i {} \;
   fi
```

ipa-mdl, 2018-12-18:
Roughly based on your solution I came up with this:

```bash
exclude=()
for p in $CLANG_FORMAT_EXCLUDE; do exclude+=(-and -not -path "$path/$p"); done
find "$path"/* $exclude -iregex '.*\.\(cpp\|hpp\|cc\|cxx\|h\|hxx\)' "${exclude[@]}"
```
* avoids `tr`, because we do not use this elsewhere -> split only by spaces
* allows blacklisting individual files
* uses relative file names (in target repository)
* Example: `CLANG_FORMAT_EXCLUDE="stomp/* industrial_core/*"`

ipa-mdl, 2019-01-21:
@Levi-Armstrong: Please elaborate on what you tried and what did not work.
I gave it a try with https://github.com/ros-industrial/industrial_moveit  and it worked as expected:

```bash
CLANG_FORMAT_EXCLUDE="stomp_moveit/src/utils/* constrained_ik/include/constrained_ik/*"

path=/tmp/industrial_moveit
git clone --depth 1 https://github.com/ros-industrial/industrial_moveit.git "$path"

exclude=()
for p in $CLANG_FORMAT_EXCLUDE; do exclude+=(-and -not -path "$path/$p"); done
find "$path"/* -iregex '.*\.\(cpp\|hpp\|cc\|cxx\|h\|hxx\)' "${exclude[@]}"
rm -rf "$path"

```

Levi-Armstrong, 2019-07-14:
@ipa-mdl Is there plans to incorporate your functionality into industrial_ci?

ipa-mdl, 2019-07-14:
> @ipa-mdl Is there plans to incorporate your functionality into industrial_ci?

I don't have time to work on this, but pull requests are welcome :)