ros-industrial/industrial_ci #346: [DOCKER_COMMIT] User has no control over whether to keep source or not in the built image

130s, 2019-01-03:
# What's the problem

With #341 merged, users can now re-use a Docker image that is built during industrial_ci pipeline. Looks like some, not all, source code that were used from the previous CI run are left on the built Docker image.

- The source from the target repo CI ran on seems to be removed.
- Source that was fetched via `wstool` seems to be left.
- `/root/ici` folder is also left, without any files.

```
$ docker pull registry.gitlab.com/foo_org/AA/AA:sampletag 
sampletag: Pulling from foo_org/AA/AA
:
90179adecea9: Pull complete 
Digest: sha256:6ec3f49b63f9bb5b08ebde0f6b01a936a6ba8b99334bef22919244622f564954
Status: Downloaded newer image for registry.gitlab.com/foo_org/AA/AA:sampletag
n130s@130s-p50:~/link/cws_PO_k/src/foo_org/AA/AA$ docker run -it registry.gitlab.com/foo_org/AA/AA:sampletag /bin/bash
root@989aac4d6dd9:/# cd catkin_ws && ll
drwxr-xr-x  2 root root 4096 Dec 10 15:15 ici/
drwxr-xr-x  3 root root 4096 Jan  2 15:25 src/
root@989aac4d6dd9:~# ll catkin_ws/src/
.rosinstall       .rosinstall.bak   faa_org/           foo_org/ 
root@989aac4d6dd9:~# ll catkin_ws/src/
.rosinstall       .rosinstall.bak   faa_org/           foo_org/ 
root@989aac4d6dd9:~# ll catkin_ws/src/foo_org/faa_org/faa_org_tools/
total 152
drwxr-xr-x 16 root root  4096 Jan  2 15:26 ./
drwxr-xr-x  8 root root  4096 Jan  2 15:26 ../
drwxr-xr-x  8 root root  4096 Jan  2 15:41 .git/
-rw-r--r--  1 root root     5 Jan  2 15:26 .gitignore
: (Files)
root@989aac4d6dd9:~# 
```

# Expected behavior

Because I can think of at least 2 following usecases for whether we want to keep source or not in the built image, it'd be great if user can choose to keep source or not.
- Keep source when the image gets built: We have a usecase where developers use the Docker image built by CI. In this case the situation described in this ticket is not bad.
- Remove source after CI pipeline finished: E.g. when shipping the image to somewhere else where you have less control, you don't want the source embedded.

ipa-mdl, 2019-01-03:
This behavior is intended.
The mentioned directories get bind-mounted and I'd like to keep it this way, because `industrial_ci` is not meant as an image build tool.
In addition bind-mounting does not need any synchronization logic and therefore is faster and less error-prone.

You could copy the code in as part of a post-process step:
1. create container
2. copy missing things in (source, credentials..)
3. commit to image

But beware: The result is still no well-formed Docker image, because it contains too many temporary files.

gavanderhoorn, 2019-01-03:
This one -- together with #347 -- seem to be a clear indication that the (mild) feature creep introduced in #341 might be a direction that `industrial_ci` would not want to go in.

@ipa-mdl wrote:
> [..] because `industrial_ci` is not meant as an image build tool. [..] The result is still no well-formed Docker image [..]

I would agree with this: `industrial_ci` is a CI driver/coordination framework, not a deployment system.

130s, 2019-01-03:
Sounds like from https://github.com/ros-industrial/industrial_ci/issues/346#issuecomment-451234777 by @ipa-mdl, some change in implementation approach would be necessary in industrial_ci in order to achieve what I suggested. And after being reminded of importance of focusing on the core feature (thanks @gavanderhoorn for the thoughts. I've never been aware of the term "feature creep", but it sounds pretty important), that kind of change would expand the scope in undesired way.

I'll have to think what I want.

Relatively minor response,

> it contains too many temporary files.

Amending this in industrial_ci wouldn't require fundamental change, would it? Does something like [what's done in `Dockerfile`s in ros-industrial/docker](https://github.com/ros-industrial/docker/blob/53f419086989684d07bf55497395c3e61c98003a/ci/melodic-bionic/Dockerfile#L16) suffice?

ipa-mdl, 2019-01-03:
> Sounds like from #346 (comment) by @ipa-mdl, some change in implementation approach would be necessary in industrial_ci in order to achieve what I suggested. 

I don't think that we should try to achive this in `industrial_ci`.
But I could imagine to come up with some tool suite that could be shared for CI and an image builder.
For example, `source_tests.sh` should get refactored properly to provide functions for setting up a workspace or installing dependencies (as part of https://github.com/ros-industrial/industrial_ci/projects/1?). 
A builder pipeline could then use these functons from a Dockerfile.

>Amending this in industrial_ci wouldn't require fundamental change, would it? 

You could remove some file in the `AFTER_SCRIPT`.

ipa-mdl, 2019-05-15:
With #361 all source file will be kept inside the container/image.

ipa-mdl, 2020-04-08:
The source files are kept now and overwritten on the next run.

closed 2020-04-08 by ipa-mdl