ros-industrial/industrial_ci #347: [DOCKER_COMMIT] environment variables carried over when a image built by ICI used again in ICI

130s, 2019-01-03:
# What's the problem

With #341 merged, users can now re-use a Docker image that is built during industrial_ci pipeline.
When using the same image on industrial_ci again, looks like the environment variables previously set during industrial_ci pipeline still remain in the image's shell.

## Attempts that didn't work
- In a job where the image is built, `unset`ting some environment variables is tried. But in the built image I still see those variables.
- Starting the built image and manually `unset`ting those variables and running `docker commit`, I see those vars are actually unset next time I start the container.

# Expected behavior

I can't think of any usecase where the environment varibles that were used to run industrial_ci remain on the shell in the image built via industrial_ci. So removing those variables before committing the image sounds reasonable to me.

ipa-mdl, 2019-01-03:
IMHO using the committed container in different environments is not a supported use case.
`rerun_ci` hashes the variables for this reason.
Even if you can remove the variables from the image/container somehow, you cannot undo the effect that easily.
Actually, I don't see any way to fix this.

Instead you might use the hashing approach as well or use the job name as a tag to prepare job-specific images.

miguelprada, 2019-02-08:
> IMHO using the committed container in different environments is not a supported use case.

This is reasonable.

> Even if you can remove the variables from the image/container somehow, you cannot undo the effect that easily.

What are you thinking of here?

We're also interested in exploring the re-use of the built images by adding some extra steps that remove unnecesary files, as discussed in #346 (e.g. `/root/catkin_ws`, `/root/ici`, `/root/src`). unset environment vars, and add suitable `CMD` and `ENTRYPOINT`. Possibly cleaning up `apt` and `rosdep` caches as well.

Are there additional cleanup steps that we should be considering in order to create deployable images? Or are there any particular reasons why you think this is a bad idea?

Of course, we could roll out a completely separate tool to prepare a deployment image, but leveraging `COMMIT_IMAGE` and post-processing the resulting image afterwards still looks like the path of least resistance.

miguelprada, 2019-02-08:
> Are there additional cleanup steps that we should be considering in order to create deployable images? Or are there any particular reasons why you think this is a bad idea?

Partially answering myself, we can't just delete `/root/catkin_ws` because everything is installed into `/root/catkin_ws/install`. Also, I believe there may be [issues](https://github.com/ros-industrial/industrial_ci/blob/8e81016601c73d191f45f3d16367157c10eedf77/industrial_ci/src/tests/source_tests.sh#L233) if the install path is changed elsewhere. We can still delete `build`, `devel` and `src`, keep `/root/catkin_ws/install` in the deployed image and source the setup files there in our entrypoint, but it's not very clean...

Any other potential issues you can think of? I think I still will give it a try.