ros-industrial/industrial_ci #349: ssh-agent error due to long path in Gitlab

jonazpiazu, 2019-01-04:
When running in a Gitlab instance, I get the following error:

```
$ .ci_config/gitlab.sh
unix_listener: path "/builds/<long_path_here>.tmp/ssh-dfVsRsvCFvXF/agent.37" too long for Unix domain socket
Could not open a connection to your authentication agent.
could not add ssh key
ERROR: Job failed: exit code 2
```

The problem: the ssh-agent creates the socket at the following address: `$TMPDIR/ssh-XXXXXXXXXX/agent.<ppid>` ; `$TMPDIR` is now specified to be ` ${CI_PROJECT_DIR}.tmp` [in the Gitlab's configuration](https://github.com/ros-industrial/industrial_ci/blob/master/.gitlab-ci.yml#L10). The problem arises when the project is part of a subgroup that is in another subgroup ... the variable `CI_PROJECT_DIR` takes then the form of `project_namespace/project_group/project_subgroup/project_subgroup/project_name` which can easily hit the path length limitation.

As a workaround, I have modified `$TMPDIR` to be `/builds/${CI_PROJECT_NAMESPACE}`. Not sure if it is the best solution though.

130s, 2019-01-07:
@jonazpiazu Thank you for the report, and suggestion!
Would you mind opening a PR to add that workaround in a documentation (then we can discuss the best solution there)?

jonazpiazu, 2019-01-11:
@130s I am not sure where do you want me to propose the change; are you suggesting to add a troubleshoot note in the [documentation](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst)? 

I think it would be better that I PR the change to the [.gitlab-ci.yml](https://github.com/ros-industrial/industrial_ci/blob/master/.gitlab-ci.yml), and then the PR itself can be used to discuss on the best solution.

130s, 2019-01-21:
> I think it would be better that I PR the change to the .gitlab-ci.yml, and then the PR itself can be used to discuss on the best solution.

+1

Then, I'd add a line in [doc/index.rst](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst) then link to `.gitlab-ci.yml` from there.

ipa-mdl, 2019-01-21:
I don't think that this needs to be fixed in the example, it is meant for the rather constrained gitlab.com shared runners.

We could add a section in the documentation when it's need and which options are available.

IMHO `TMPDIR` should point to a dedicated folder which cannot interfere with the rest.
So `/builds/${CI_PROJECT_NAMESPACE}` is not an optiomal choice..