ros-industrial/industrial_ci #350: [gitlab-ci] Adding SSH_PRIVATE_KEY and SSH_SERVER_HOSTKEYS breakes the CI

tgaspar, 2019-01-10:
Hi everyone,

I followed the instructions written in the [documentation](https://github.com/ros-industrial/industrial_ci/blob/master/doc/index.rst#gitlab-ci-access-to-private-repositories) regarding adding private repositories. I added my SSH_PRIVATE_KEY and SSH_SERVER_HOSTKEYS as instructed and as far as I know the commands described [here](https://docs.gitlab.com/ee/ci/ssh_keys/README.html#ssh-keys-when-using-the-docker-executor) are ran.

However, the CI fails with the following error (I also added the line describing which job was the last successful one for clarity):

```console
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
Function prepare_docker_image returned with code '0' after 1 min 42 sec 

tar: unrecognized option: owner=0
BusyBox v1.28.4 (2018-12-06 15:13:21 UTC) multi-call binary.

Usage: tar c|x|t [-ZzJjahmvO] [-f TARFILE] [-C DIR] [-T FILE] [-X FILE] [--exclude PATTERN]... [FILE]...

Create, extract, or list files from a tar file

Operation:
	c	Create
	x	Extract
	t	List
	-f FILE	Name of TARFILE ('-' for stdin/out)
	-C DIR	Change to DIR before operation
	-v	Verbose
	-Z	(De)compress using compress
	-z	(De)compress using gzip
	-J	(De)compress using xz
	-j	(De)compress using bzip2
	-a	(De)compress using lzma
	-O	Extract to stdout
	-h	Follow symlinks
	-m	Don't restore mtime
	-T FILE	File with names to include
	-X FILE	File with glob patterns to exclude
	--exclude PATTERN	Glob pattern to exclude
ERROR: Job failed: exit code 1
```

If I got that right, this happens during the execution of [docker.sh](https://github.com/ros-industrial/industrial_ci/blob/master/industrial_ci/src/docker.sh). More precisely, during the execution of the `docker_cp` function for `~/.ssh` deirectory:
```bash
function docker_cp {
  set -o pipefail
  tar --numeric-owner --owner=${docker_uid:-root} --group=${docker_gid:-root} -c -f - -C "$(dirname $1)" "$(basename $1)" | docker cp - $2
  set +o pipefail
}
```
I am using a docker in docker setup with the `.gitlab-ci.yml` looking roughly like this:
```yaml
image: docker:git
variables:
  DOCKER_DRIVER: "overlay2" # recommended for docker-in-docker
services:
  - docker:dind # enable docker-in-docker
before_script:
  # Clone the industrial_ci package
  - apk add --update bash coreutils # install travis depedencies
  - git clone https://github.com/ros-industrial/industrial_ci.git .ci_config
 
kinetic:
  script: 
    - .ci_config/gitlab.sh
  variables:
    ROS_DISTRO: "kinetic"
    CATKIN_LINT: "true"
    OPT_VI: "-vi"
    UPSTREAM_WORKSPACE: "file" # Tell them to install stuff from a .rosinstall file
    ROSINSTALL_FILENAME: "<MY_PKG_NAME>/rosdep.rosinstall"
```

I am totally stuck here and some help would be greatly appreciated.

miguelprada, 2019-01-10:
You also need to update `tar` in the dependencies installed via `apk`.

ipa-mdl, 2019-01-10:
> You also need to update tar in the dependencies installed via apk.

Examples: [README](https://github.com/ros-industrial/industrial_ci#for-gitlab-ci), [.gitlab-ci.yml](https://github.com/ros-industrial/industrial_ci/blob/master/.gitlab-ci.yml#L17)