ros-industrial/industrial_ci #353: Custom rosdeps?

reinzor, 2019-01-29:
Is there a way to specify custom rosdep sources? I cannot find it in the documentation or the issues. Thanks

130s, 2019-02-02:
Probably not possible as of now. Only customization possible for `rosdep` so far is [ROSDEP_SKIP_KEYS](https://github.com/ros-industrial/industrial_ci/blob/0cd9b4adc908c8dbb00cf04d2361e60a99ccaa41/industrial_ci/src/tests/source_tests.sh#L161).

I've never tried though, `rosdep` seems to allow passing custom rosdep source.
```
$ rosdep --help                                                                                                                                                                      
:
-c SOURCES_CACHE_DIR, --sources-cache-dir=SOURCES_CACHE_DIR
                        Override /home/useraccount/.ros/rosdep/sources.cache

$ apt-cache policy python-rosdep
python-rosdep:
  Installed: 0.15.0-1
```
If you can open a pull request that'll be great.

miguelprada, 2019-02-04:
We've been exploring different ways to add `rosdep` sources and have experimented with:

- adding them "manually" with a script executed in `BEFORE_SCRIPT`
- adding them to a docker image, which is selected with `DOCKER_IMAGE`

In both cases, we also set-up custom `apt` sources, and so far both approaches have worked fine.

I think we faced some issues when trying to pass the custom image as `DOCKER_BASE_IMAGE`, but it wasn't me setting this up and I can't remember exactly what went wrong.

VictorLamoine, 2019-02-14:
When I release closed source software to clients I do my releases locally, this requires rosdep keys to be added so I have created scripts to help me do the job:
https://gitlab.com/VictorLamoine/bloom_local_release

The `append_rosdep_key`, `clear_rosdep_keys` will help you do the job. There are some comments in the script so you should be able to understand how it works.

This could be a good starting point for you, I hope it helps!