ros-industrial/industrial_ci #356: CATKIN_LINT and PRE_RELEASE cannot be set at the same time?

musamarcusso, 2019-02-28:
Hi,
I just noticed that if the `PRE_RELEASE` is set to true, `catkin_lint` will not be called when `CATKIN_LINT` is also true. Is it possible to have both options in one build or should I have one extra build without the `PRE_RELEASE` option to run it?

ipa-mdl, 2019-02-28:
For now the `CATKIN_LINT ` option can only be used as part of the regular source tests.
The reason behind is that the dependencies get resolved first, which works a little bit different for the pre-release tests (from `ros_buildfarm`) and is hard to integrate.

I could imagine to factor the `catkin_lint` part out into an independent test, which can be run without the other source tests.

musamarcusso, 2019-02-28:
Thanks for the clarification, @ipa-mdl. I will change my Travis configuration to have this set up.

closed 2019-02-28 by musamarcusso